mp-se/gravitymon #45: show the correct device.local name to connect to in initial setup

Josverl, 2022-04-03:
## Issue 
On initial setup I found that the IP address generated by the interface was incorrect ( 192.168.4.1
and it was quite hard to find either : 
- the mdns name (requires installation of mdns tools) 
- or the device IP address, 
  - apparently no host name is associated with the DHCP request 
  - as the DHCP lease is released quite quickly , the router does not show the client in the lease table 
- I ended up reading the serial debug to find the IP address 

## Expected behaviour: 
- the mdsn name to be `<devicename>.local `  as documented : https://mp-se.github.io/gravitymon/configuration.html
I think tha in order to run multiple devices on the the same network you have added a few characters to the mdns name, which works quite well im my own projects,  however that only works if this is also shows to an end user.

## Proposed  fix :
- the used mdsns name to be shown and linked in the setup UX as [`http://<devicename>ABCDEF.local`](http://devicenameABCDEF.local)

mp-se, 2022-04-03:
Dont really understand what the issue is. When doing the initial wifi setup the ip for the device is hardcoded to 192.168.4.1 which is also stated in the documentation. 

MDNS would not be active until it joins the network after WIFI setup. If you are using Windows you would need to install 3rd party tools for this support. 

You have a point regarding the MDNS name since I make this unique. I would be good to have this visible on the WIFI setup page.

mp-se, 2022-04-04:
Added device name to wifi setup page

closed 2022-04-04 by mp-se