mp-se/gravitymon #46: Temp Sensor Adj Problem

StodgyWaif, 2022-04-03:
Hello.
I changed my OTA Base URL to:
https://github.com/mp-se/gravitymon/tree/master/bin/
(Not sure if this is correct or not)

And the Temp Sensor Adj value automatically goes to -17.77778.
I tried changing it to 0 but it just goes back to that value.
Even after removing the OTA value I cannot set the Temp Sensor value back to 0.

I tried in both Chrome and Firefox.
This is on the 0.8.0 version.

mp-se, 2022-04-03:
Good find. This is another bug. Internally I use C as the standard format and I missed to convert this parameter if F was used. 

It should work to use the github URL for the OTA base, but I have not yet tested it. Its a good suggestion.

mp-se, 2022-04-04:
Fixed in dev

closed 2022-04-04 by mp-se