mp-se/gravitymon #47: Wrong over push temp when using F as temp

mp-se, 2022-04-03:
Hello.
I just installed GravityMon and it seems to be working well but I noticed the temperature recorded in Fermentrack is 30 degrees F or more higher than what the GravityMon web server page shows. Is this a push configuration error?
Thanks for any advice or information.

EDIT:
Okay I just figured out that GravityMon is detecting the temperature as 68F but it's sending it to Fermentrack as 68C.
Any idea how I can correct this?

mp-se, 2022-04-03:
It looks like that the ispindle format always should be in Celsius. So this is a bug. If you change the format template to the following it should work:

{"name" : "${mdns}", "ID": "${id}", "token" : "${token}", "interval": ${sleep-interval}, "temperature": ${temp-c}, "temp-units": "${temp-unit}", "gravity": ${gravity}, "angle": ${angle}, "battery": ${battery}, "rssi": ${rssi}, "corr-gravity": ${corr-gravity}, "gravity-unit": "${gravity-unit}", "run-time": ${run-time} }

StodgyWaif, 2022-04-03:
As a work around I entered in Format Editor:
"temperature": ${temp-c},

mp-se, 2022-04-03:
Looks like I had a spelling error in the format defintion. The temp format should be "temp_units" and not "temp-units"... If you change the format to this it should work. Since fermentrack could not find the format type it defaults to celsius

{"name" : "${mdns}", "ID": "${id}", "token" : "${token}", "interval": ${sleep-interval}, "temperature": ${temp}, "temp_units": "${temp-unit}", "gravity": ${gravity}, "angle": ${angle}, "battery": ${battery}, "rssi": ${rssi}, "corr-gravity": ${corr-gravity}, "gravity-unit": "${gravity-unit}", "run-time": ${run-time} }

mp-se, 2022-04-04:
Fixed in dev

closed 2022-04-04 by mp-se