mp-se/gravitymon #60: Timing Settings and  Integration with Brewfather Version 0.9

Hwerow, 2022-04-25:
Congratulations - a great project with lots of features.
1 Please consider having different timing settings for  HTTP 1 say for Brewfather 900s and HTTP2 MQTT 180s this would allow good flexibility in use.

2 Brewfather I was not able to replicate the same information that I get from my iSpindel using gravitymon. This may of course be user error! But it would be good to have the same data built into gravitymon as is sent via iSpindel. Brewfather expects Plato and the iSpindel name has to include [SG] - not sure how this might affect matters? 
[image]

The example shown was using the iSpindel setting the Custom stream showed less information.
Cheers

mp-se, 2022-04-25:
On your suggestion 1) It would be possible to have different timings for each endpoint, but since the device only have one timer it's probably best to have a standard sleep interval and then set that a target would send ever X wakeup cycle. For instance 300s as default and sending to http 1 every 3rd wakeup. The downside is that this would require a lot of writing to the flash memory which might impact the lifecycle of the esp. But I will investigate this (I will split this into it's own issue). 

Another option is to have some intermediate software to act as a proxy where data is sent to external endpoints. From a battery life standpoint this would probably be the best option. I know that fermentrack can be used to relay data to other services. Im considering doing some type of web application that can manage the devices and be use for testing and as a proxy. 

2) What option did you choose for brewfather (there are 2). You can use http1/http2 with the brewfather iSpindle endpoint or the brewfather option in the UI (which is the custom endpoint)? I remember having this disussion with another user and the ispindle enpoint worked better than the custom stream. I will do some more testing on brewfather to see how the data is transformed.

According to the docs you will need to add [SG] as a prefix on the device for brewfather to do conversion correctly. You can do this in the device name or add it to the format template.

Hwerow, 2022-04-25:
1 It would be neat if 2 different timing cycles could be implemented. iSpindHub might be  an intermediate option  https://github.com/ZeSlammy/iSpindHub

2 I tried both 
- the Brewfather UI custom - did not send all the equivalent data to an iSpindel, the example data in your doco seems incomplete compared to what an iSpindel sends
- the iSpindel http1, which is what the screen capture shows, incomplete data. I thought that it might be that the data being sent was incompatible with what Brewfather was expecting?

mp-se, 2022-04-25:
Thanks for the update. Will do some more testing to figure out what is happening later today.

mp-se, 2022-04-25:
First item has been split into its own issue #61

mp-se, 2022-04-25:
I've been looking into the brewfather integration and the name of the device needs to end with [SG] or brewfather will convert the  value from plato to SG. If the custom endpoint is used then the gravity format is specified so the name does not have to change. 

If you dont want to change the name of the device you can also customize the format template to include the [SG], like this:

"name" : "${mdns}[SG]"

I will take an action to update the documentation to make it clear on this point.

Hwerow, 2022-04-26:
OK. For the record I have now used http 1 with the iSpindel id for Brewfather and an added [SG]  with the template as below. For some reason rssi is not shown in Brewfather? not a huge deal - just strange
[image]
[image]

mp-se, 2022-04-26:
It looks like there is a bug in brewfather, if you change "rssi" to "RSSI" the wifi strenght will show up. Variables in json documents should not be case sensitive.

Hwerow, 2022-04-26:
Thanks - looks like my connection to Brewfather is now sorted.

mp-se, 2022-04-26:
Great news. I will update the docs on this topic so the next user dont have these issues.

mp-se, 2022-04-26:
Notes! Do these changes:

- Update documentation to clarify [SG] prefix in brewfather integration
- Add pre-defined format template that sends data using plato to avoid having to add prefix
- Add  pre-defined iSpindle legay format to bypass the issue with require json parameters to be in upper case (RSSI and ID) params are affected.

closed 2022-04-26 by mp-se

mp-se, 2022-04-29:
@Hwerow I've publised a beta version that you can try out. The docs are not yet updated, you can change the frequency of the push targets under configuration - advanced. You need to click the checkbox first to enable these settings. 

0 means every execution. 1 means every seconds  etc. 

https://github.com/mp-se/gravitymon/releases/tag/v1.0.0-beta1