mp-se/gravitymon #62: Traveling brewer - backup WIFI configuration

jsbahnsen, 2022-04-26:
Hi,

Thanks for your great work. 

Would it be possible to implement a feature that changes to an alternative WIFI configuration e.g. when 3 consecutive connection on wake up attempts have failed - then trying the backup and using this as primary if connection is successful. My use case would be brewing and fermenting at different locations or for instance cold crashing outdoors where another SSID is covering the area.

KR Jesper

mp-se, 2022-04-26:
It's a good idea, it could also be possible to scan for which networks that are available and connect to one that is best suited. Will need to do some experimentation on what is the best option from a performance / battery perspective.

mp-se, 2022-04-27:
i've done some testing on the wifi connection/scanning. I see two options for implementing this. One option is to try to connect and in case of failure switch the to the secondary wifi settings. The other option is to scan for the strongest wifi network (in case more than one is defined) and then connect to the strongest. 

A failed connect takes 20 seconds and a network scan takes around 2 seconds. 2 seconds is the same time to read the sensor and do a normal wifi connect to it will in practice reduce the battery life with 50%.  But if  we need to spend 20s at every cycle then the battery will drain much faster.... 

Both options have pros/cons so I'm not sure what option is the best.

mp-se, 2022-04-28:
Add so that the current used network is shown on the index page. Will focus on implementing the 2:d option where the second SSID is used as fallback in case of errors.

closed 2022-04-28 by mp-se

jsbahnsen, 2022-04-29:
Great, thanks. 
I agree that this is the best option of the two. If possible it may be more efficient if the secondary SSID is changed to be primary after failure and connection to secondary. So next wake up it will attempt secondary SSID and then primary.

I hope this makes sense
KR Jesper

mp-se, 2022-04-29:
Thats exactly how I implemented it, i plan to do an beta release of 1.0 this weekends so I will tag you once that is done. Id appreciate some feedback on the feature.

jsbahnsen, 2022-04-29:
Will be glad to help

mp-se, 2022-04-29:
@jsbahnsen  I've publised a beta version that you can try out. The docs are not yet updated, you can now set two ssid in the wifi setup page.

https://github.com/mp-se/gravitymon/releases/tag/v1.0.0-beta1