mp-se/gravitymon #64: Temperature not showing up at UBIDots when using standard template.

mp-se, 2022-04-27:
Report from homebrewtalk.com check if this is a bug

eightheads, 2022-05-25:
I'm running into the same with a new build and v1.0.0 firmware

eightheads, 2022-05-25:
Included are screenshots of my Format Template for Ubidots along with my Device activity on the Ubidots side.

[image]

[image]

mp-se, 2022-05-25:
1) You are not using the latest updated format template. Clear the field and press save (or copy if from the list). This way you are using the latest format template.
2) I noticed that only 10 out of the 13 variables show up so there might be a limitation on how many values you can post to ubidots per device (have not found anything in the docs about this). I would suggest that you update the format to contain only the value that you want to show up in ubidots.

eightheads, 2022-05-25:
> 1. You are not using the latest updated format template. Clear the field and press save (or copy if from the list). This way you are using the latest format template.

This was from selecting iSpindel (Post) in the drop-down and clicking Copy format. This is a new install as I have not used GravityMon before, and this was on a new D1 Mini and new iSpindel build. I have yet to have it display temperature in Ubidots with GravityMon. I will try again when I get home tonight. Can you also maybe reply with what the format should be?

> 2. I noticed that only 10 out of the 13 variables show up so there might be a limitation on how many values you can post to ubidots per device (have not found anything in the docs about this). I would suggest that you update the format to contain only the value that you want to show up in ubidots.

After testing with resetting the Push format, I will then test with reducing to 10 fields or less.

mp-se, 2022-05-25:
You are correct, there is a bug in the "copy format" template. I will fix that. You can try this format that should contain most of the data that you need; 

{
 "name" : "${mdns}",
 "interval": ${sleep-interval},
 "temperature": ${temp},
 "temp_units": "${temp-unit}",
 "gravity": ${gravity},
 "angle": ${angle},
 "battery": ${battery},
 "RSSI": ${rssi}
}

eightheads, 2022-05-25:
> You are correct, there is a bug in the "copy format" template. I will fix that. You can try this format that should contain most of the data that you need;
> 
> { "name" : "${mdns}", "interval": ${sleep-interval}, "temperature": ${temp}, "temp_units": "${temp-unit}", "gravity": ${gravity}, "angle": ${angle}, "battery": ${battery}, "RSSI": ${rssi} }
> 
> You can also try the ubidots format from the list, it contains a few of the params.

Can you check the BrewFather "copy format" as well? It does not seem to be sending all parameters either.

mp-se, 2022-05-25:
Did you use it with the brewfather custom stream or ispindle?

eightheads, 2022-05-25:
> Did you use it with the brewfather custom stream or ispindle?

The Brewfather stream

mp-se, 2022-05-25:
It works for me, it needs to be the custom stream endpoint. The normal iSpindle endpoint uses the standard iSpindle format. 

[image]

eightheads, 2022-05-25:
> It works for me, it needs to be the custom stream endpoint. The normal iSpindle endpoint uses the standard iSpindle format.
> 
> [image]

Sorry, correct, I'm using the iSpindle endpoint and copied the Brewfather format. So, I should instead use the "Option 2 - Custom Stream" to use the Brewfather format? I'll change that out as well when I'm back home.

eightheads, 2022-05-25:
@mp-se what are the Pros/Cons to the iSpindle endpoint and Custom endpoint?

mp-se, 2022-05-25:
No problems, i will see if i can make the documentation more clear

mp-se, 2022-05-25:
The custom stream is more generic and have some values that is not in the iSpindle format but lack some of the data from the iSpindle. I would recommend the iSpindle format since that is better for this project. I was using the custom endpoint for another project doing pressure measurement so I just reused some of the code from that project. I've kept it for backwards compatibility since it has been used in previous versions.

eightheads, 2022-05-26:
Using your suggested format above and after removing the device from Ubidots and letting it add itself back I'm now getting Temperature properly, but not receiving Name or Temp Units for some reason.
Note: this is being tested in water and I have not run through the formula calibration yet.

[image]

mp-se, 2022-05-26:
I'm not a ubidots user but from what I can understand it treats every param as a number. thats why the name is not shown. For the text to be shown you need to create a custom device type and tell ubidots which variables are text and what is numbers. but in order to do that you need a paid subscription which I dont have. You can check this article: https://help.ubidots.com/en/articles/2129204-device-types (that seams to validate that there is a limit of 10 variables per device)

eightheads, 2022-05-26:
Thanks for the help. It seems to be receiving data reliably now. Brewfather is working now as well now that I changed the format back and reduced to 10 variables. A simple fix of the iSpindle format template should fix that. Thanks again for all your work on this project!

mp-se, 2022-05-26:
As i suspected, good to have this confirmed, I'll close this issue now. If you have more issues please raise a new issue.

closed 2022-05-26 by mp-se