mp-se/gravitymon #71: Comments on 1.0.0-beta1 MQTT and Advanced Settings

Hwerow, 2022-04-30:
1 Setting up MQTT - was working OK in 0.9 with this formula
gm:{"name":"${mdns}", "ID":"${id}", "token":"gravmon", "temperature": ${temp}, "temp_units": "${temp-unit}", "gravity":${gravity},
"angle": ${angle}, "battery":${battery}, "rssi": ${rssi}}
When you input the MQTT format and test all looks good. 
[image]
As soon as you press save the information disappears but is says successfully stored
[image]
Hit save
[image]
Error message
PUSH: MQTT push on gm failed error=-5
No message received

2 Advanced configuration
The enable Advanced configuration check box seems to be working back to front.  You have to uncheck it to activate the save button
[image]
[image]

mp-se, 2022-04-30:
This is strange, I tested your seettings on my dev device and it works there. On this page I have some javascript code that does some handling in the background so perhaps it's an script problem on the browser. 

1. What OS & Browser are you using?
2. Can you open the javascript console (should be under developer tools in the menu)
3. Run the test and see if you get any errors (you should see the data sent between the browser and device in that console)

On the advanced settings, it's intentional that the fields/save button is disabled by default. These settings can cause issues to I wanted to make the users aware that this should be used with caution. Do yo think it's a anoyance ?

Hwerow, 2022-04-30:
1 I am using Firefox and Win10
2 3 and 4 I will do tomorrow as I have reverted to 0.9

The advanced setting as it is is confusing - perhaps an "Are you sure" second step would be better?

PS Fahrenheit is the correct spelling 
cheers

Hwerow, 2022-05-04:
Using the same details as at the top of this issue
2 & 3
I have tested using 1.0.0beta with Firefox and Edge browsers using iSpindel (MQTT)
**Saving Using 1.0.0beta Edge** - nothing shown 
[image]
**Saving Using 1.0.0beta Firefox** - nothing shown
[image]

Both show  > mqtt: null and nothing is displayed on the screen but it would appear to be saved

4 I can't see any file not saved error. 

If I now select Copy format this is what I get
[image]

Then Select Test
[image]

Then press Save and nothing shown
[image]

The error log shows this - however,  there is no clear button to know if this is stale info or not. The device that I am using to monitor the MQTT is not receiving messages.

[image]

So I am stuck. Please  let me know if there is anything else I should try.

mp-se, 2022-05-04:
From the posts it looks like if fails to save the format on the device for some reason. If even the standard template does not work one possibility is that the file system has been corrupt for some reason. I will do some tests later today (its morning here in sweden). The code has not changed in that part since 0.9 but i might need to do some more error checking.

 You could try to do a full wipe of the device to format the file system partition and see if that works.

Hwerow, 2022-05-04:
OK I will try uploading the bin via cable and wipe the data. I updated OTA to GM 0.9 using the iSpindel 7.10 firmware then went from 0.9 - 1.0.0 using GM so there may be something lurking?

mp-se, 2022-05-04:
Ive checked the code and you should get an error if the file cannot be saved. So it's likley that the function i use to do an URL encode of the string crashes and returns null. If this is the case you should get the same issue when trying to save that format template to one of the other endpoints. 

If the file is missing or empty you should get the default format template back. So I must have not used the exact string since I could not replicate the issue. Will do some more testing.

mp-se, 2022-05-04:
> OK I will try uploading the bin via cable and wipe the data. I updated OTA to GM 0.9 using the iSpindel 7.10 firmware then went from 0.9 - 1.0.0 using GM so there may be something lurking?

From what I remember iSpindle is using the old SPIFFS file system so GM should format that when starting up. But I can try that   as well.

mp-se, 2022-05-04:
If you have the same issue after reflashing could you copy a section that looks like this from the javascript console ? That would contain the exact data that is to be stored. With that I can hopefully replicate the issue in that case.  Look for id=...mqtt=....

`` 
id=66a1b0&mqtt=gm%3A%7B%22name%22%3A%22%24%7Bmdns%7D%22%2C%20%22ID%22%3A%22%24%7Bid%7D%22%2C%20%22token%22%3A%22gravmon%22%2C%20%22temperature%22%3A%20%24%7Btemp%7D%2C%20%22temp_units%22%3A%20%22%24%7Btemp-unit%7D%22%2C%20%22gravity%22%3A%24%7Bgravity%7D%2C%22angle%22%3A%20%24%7Bangle%7D%2C%20%22battery%22%3A%24%7Bbattery%7D%2C%20%22rssi%22%3A%20%24%7Brssi%7D%7D``

Hwerow, 2022-05-05:
Right, I have reflashed and still have the problem. 
So I paste in gm:{"name":"${mdns}", "ID":"${id}", "temperature": ${temp}, "gravity":${gravity},
"angle": ${angle}, "battery":${battery}, "rssi": ${rssi}}

[image]
Pressing  save blanks out the information and nothing comes up on the console mqtt  = null

Now if I select copy format
[image]
and paste over the data shown with the details above 
The console shows this
id=389c58&mqtt=gm%3A%7B%22name%22%3A%22%24%7Bmdns%7D%22%2C%20%22ID%22%3A%22%24%7Bid%7D%22%2C%20%22temperature%22%3A%20%24%7Btemp%7D%2C%20%22gravity%22%3A%24%7Bgravity%7D%2C%22angle%22%3A%20%24%7Bangle%7D%2C%20%22battery%22%3A%24%7Bbattery%7D%2C%20%22rssi%22%3A%20%24%7Brssi%7D%7D

This may or may not be related
The MQTT data has been set to send every 180 secs 3 mins - yet my device only receives a single message. 
I have used the advanced settings to skip 5 = 900 secs 15 minutes for Brewfather which is updating as expected

Hwerow, 2022-05-05:
When I press test after entering the details - not a javascript object
[image]

mp-se, 2022-05-05:
Thanks, I will test this tonight. I hope this will help me to replicate the issue and find what's wrong.

mp-se, 2022-05-05:
> When I press test after entering the details - not a javascript object [image]

The error message is normal (it should however say - not a json object) the format for used for MQTT is no a valid json format so the error is ok.

Unfortunately I cant replicate the error, if I use the exact same payload that fails for you it works for me. So a final thing to try is to hook the device to a serial monitor and see if there are any errors or clues that I can use to figure out what is happening. There is some info on that topic in the docs.

Hwerow, 2022-05-06:
OK, I will give the serial monitor a try on the weekend.

mp-se, 2022-05-06:
I've just published beta2 so you could try that one. Ive added a feature to get some more details from the device for easier troubleshooting.

Hwerow, 2022-05-06:
OK, I will give it a whirl.

Hwerow, 2022-05-09:
Using 1.0.0 beta-2   
Serial Monitor log Interval set for 180 secs using Serial Port Monitoring Tool
I removed the actual ssid and IP address details - the rest is unchanged.
The log shows 4 successful publishes then errors eg 6971 E: PUSH: MQTT push on gm failed error=-5  Which is Connection refused, not authorized

I have a BrewPiLess sending MQTT data to the same broker and that is working so I wondered if there is something different in the gravitymonitor MQTT sending setup?
[220506_gm_beta2_device logs_ed.txt](https://github.com/mp-se/gravitymon/files/8648771/220506_gm_beta2_device.logs_ed.txt)

mp-se, 2022-05-09:
Thanks, from what i can see it looks like the network connection closes before the push is xompleted, and I have an idea of what can cause it. Will do a build later for you to test out and post in this thread

Hwerow, 2022-05-10:
I have been running the updated firmware using MQTT for a few hours now, with a 180 sec interval, and it has maintained connection to the broker.  So I think you have solved it. Thanks

mp-se, 2022-05-10:
> I have been running the updated firmware using MQTT for a few hours now, with a 180 sec interval, and it has maintained connection to the broker. So I think you have solved it. Thanks

Excellent news, I will create a new beta based on this input. If no more bugs are found this will be the 1.0 release.

closed 2022-05-10 by mp-se