mp-se/gravitymon #79: Tilt angle v SG at #1 graph calculation issue

Harrym1b, 2022-05-16:
Version is 1.0.0 (..2d7ad4)
These are the values I attempted to use - 
1   23.52 1.000 <<<<< 
2   26.47 1.009
3   29.87 1.014
4   33.43 1.027
5   38.16 1.037
6   40.60 1.042
7   45.85 1.051
8   50.12 1.060
9   56.55 1.073
10 59.078 1.078

Only by setting 1 to 0.00deg can I get it to calculate the graph successfully

mp-se, 2022-05-16:
If you used one of the 1.0 beta versions the first value might have been corrupted, this will cause the formula to be rejected. To confirm this can you use the button on the first page (Gather support information) and paste the section that starts with;    "formula-calculation-data": {
      "a1":25,
      "a10":0,
      ....
      "g1":1,
      "g10":1
   },

I suspect that the "g1" value states "0" and not "1". 

If this is the case the easiest way is to do a factory reset by entering this URL in the device:

http://<yourip>/api/factory?id=<device id from the index page>

It should keep the wifi settings.

An other option is to not use the first field and set that angle to 0. In that case it should be ignored and not used. I plan to fix this in the next version.

Harrym1b, 2022-05-16:
I have done a reinstall from a fresh copy of V1.0 and again to set #1 with the same result. I also tried the Gather Support and there is no mention of  "formula-calculation-data" in it. I have copied all of it below...

 {
  "mdns": "gravitymonc75a69",
  "id": "c75a69",
  "sleep-interval": 60,
  "temp-format": "C",
  "gravity-format": "G",
  "gravity-temp-adjustment": true,
  "voltage-factor": 1.64,
  "platform": "esp8266",
  "ble": "",
  "gyro-temp": false,
  "gyro-calibration-data": {
    "ax": -4648,
    "ay": 1849,
    "az": 1342,
    "gx": 47,
    "gy": 7,
    "gz": 41
  },
  "temp-adjustment-value": 0,
  "runtime-average": 17.6,
  "rssi": -71,
  "app-ver": "1.0.0",
  "app-build": "..2d7ad4",
  "formats": {
    "id": "c75a69",
    "http-1": "%7B%22name%22%20%3A%20%22%24%7Bmdns%7D%22%2C%20%22ID%22%3A%20%22%24%7Bid%7D%22%2C%20%22token%22%20%3A%20%22%24%7Btoken%7D%22%2C%20%22interval%22%3A%20%24%7Bsleep%2Dinterval%7D%2C%20%22temperature%22%3A%20%24%7Btemp%7D%2C%20%22temp%5Funits%22%3A%20%22%24%7Btemp%2Dunit%7D%22%2C%20%22gravity%22%3A%20%24%7Bgravity%7D%2C%20%22angle%22%3A%20%24%7Bangle%7D%2C%20%22battery%22%3A%20%24%7Bbattery%7D%2C%20%22RSSI%22%3A%20%24%7Brssi%7D%2C%20%22corr%2Dgravity%22%3A%20%24%7Bcorr%2Dgravity%7D%2C%20%22gravity%2Dunit%22%3A%20%22%24%7Bgravity%2Dunit%7D%22%2C%20%22run%2Dtime%22%3A%20%24%7Brun%2Dtime%7D%20%7D",
    "http-2": "%7B%22name%22%20%3A%20%22%24%7Bmdns%7D%22%2C%20%22ID%22%3A%20%22%24%7Bid%7D%22%2C%20%22token%22%20%3A%20%22%24%7Btoken%7D%22%2C%20%22interval%22%3A%20%24%7Bsleep%2Dinterval%7D%2C%20%22temperature%22%3A%20%24%7Btemp%7D%2C%20%22temp%5Funits%22%3A%20%22%24%7Btemp%2Dunit%7D%22%2C%20%22gravity%22%3A%20%24%7Bgravity%7D%2C%20%22angle%22%3A%20%24%7Bangle%7D%2C%20%22battery%22%3A%20%24%7Bbattery%7D%2C%20%22RSSI%22%3A%20%24%7Brssi%7D%2C%20%22corr%2Dgravity%22%3A%20%24%7Bcorr%2Dgravity%7D%2C%20%22gravity%2Dunit%22%3A%20%22%24%7Bgravity%2Dunit%7D%22%2C%20%22run%2Dtime%22%3A%20%24%7Brun%2Dtime%7D%20%7D",
    "http-3": "%3Fname%3D%24%7Bmdns%7D%26id%3D%24%7Bid%7D%26token%3D%24%7Btoken2%7D%26interval%3D%24%7Bsleep%2Dinterval%7D%26temperature%3D%24%7Btemp%7D%26temp%2Dunits%3D%24%7Btemp%2Dunit%7D%26gravity%3D%24%7Bgravity%7D%26angle%3D%24%7Bangle%7D%26battery%3D%24%7Bbattery%7D%26rssi%3D%24%7Brssi%7D%26corr%2Dgravity%3D%24%7Bcorr%2Dgravity%7D%26gravity%2Dunit%3D%24%7Bgravity%2Dunit%7D%26run%2Dtime%3D%24%7Brun%2Dtime%7D",
    "influxdb": "measurement%2Chost%3D%24%7Bmdns%7D%2Cdevice%3D%24%7Bid%7D%2Ctemp%2Dformat%3D%24%7Btemp%2Dunit%7D%2Cgravity%2Dformat%3D%24%7Bgravity%2Dunit%7D%20gravity%3D%24%7Bgravity%7D%2Ccorr%2Dgravity%3D%24%7Bcorr%2Dgravity%7D%2Cangle%3D%24%7Bangle%7D%2Ctemp%3D%24%7Btemp%7D%2Cbattery%3D%24%7Bbattery%7D%2Crssi%3D%24%7Brssi%7D%0A",
    "mqtt": "ispindel%2F%24%7Bmdns%7D%2Ftilt%3A%24%7Bangle%7D%7Cispindel%2F%24%7Bmdns%7D%2Ftemperature%3A%24%7Btemp%7D%7Cispindel%2F%24%7Bmdns%7D%2Ftemp%5Funits%3A%24%7Btemp%2Dunit%7D%7Cispindel%2F%24%7Bmdns%7D%2Fbattery%3A%24%7Bbattery%7D%7Cispindel%2F%24%7Bmdns%7D%2Fgravity%3A%24%7Bgravity%7D%7Cispindel%2F%24%7Bmdns%7D%2Finterval%3A%24%7Bsleep%2Dinterval%7D%7Cispindel%2F%24%7Bmdns%7D%2FRSSI%3A%24%7Brssi%7D%7C"
  },
  "advanced": {
    "gyro-read-count": 50,
    "gyro-moving-threashold": 500,
    "formula-max-deviation": 1.6,
    "wifi-portal-timeout": 120,
    "wifi-connect-timeout": 20,
    "push-timeout": 10,
    "formula-calibration-temp": 20,
    "int-http1": 0,
    "int-http2": 0,
    "int-http3": 0,
    "int-influx": 0,
    "int-mqtt": 0,
    "tempsensor-resolution": 9,
    "ignore-low-angles": false
  },
  "files": {
    "index": false,
    "config": false,
    "calibration": false,
    "format": false,
    "about": false,
    "test": false,
    "files": [
      {
        "file-name": "drd.dat",
        "file-size": 4
      },
      {
        "file-name": "error.log",
        "file-size": 705
      },
      {
        "file-name": "gravitymon.json",
        "file-size": 1079
      }
    ]
  },
  "log": [
    "PUSH: HTTP get failed response=-1\r",
    "PUSH: HTTP post failed response=-1 (http)\r",
    "PUSH: HTTP get failed response=-1\r",
    "PUSH: HTTP post failed response=-1 (http)\r",
    "PUSH: HTTP get failed response=-1\r",
    "PUSH: HTTP post failed response=-1 (http)\r",
    "PUSH: HTTP get failed response=-1\r",
    "PUSH: HTTP post failed response=-1 (http)\r",
    "PUSH: HTTP get failed response=-1\r",
    "PUSH: HTTP post failed response=-1 (http)\r",
    "PUSH: HTTP get failed response=-1\r",
    "PUSH: HTTP post failed response=-1 (http)\r",
    "CALC: Error validating created formula. Deviation to large, formula rejected.\r",
    "CALC: Error validating created formula. Deviation to large, formula rejected.\r",
    "CALC: Error validating created formula. Deviation to large, formula rejected.\r",
    ""
  ]
}

Harrym1b, 2022-05-16:
Thanks, I will give that a try.

Harrym1b, 2022-05-16:
Increasing the value to 10 does work, thanks..

mp-se, 2022-05-17:
Great, i will see if i can make this feature better in the future