mp-se/gravitymon #80: Gyro moving error: Gravity/Angle both only display Gyro moving

eightheads, 2022-05-28:
**Describe the bug**
I re-flashed(wiped flash) one of my previously working iSpindle running GravityMon, went through initial setup, reset power, then went back to calibrate. Gravity and Angle are only displaying "Gyro moving" now and cannot calibrate. I have tried setting the iSpindle on different surfaces including my floor to make sure it is not moving at all, but only see Gyro moving messages. I have tried calibrating multiple times but seems to lockup with solid blue light on D1 Mini. I have let it sit for 5+ minutes with no change. 

Note: This device was previously running GravityMon successfully but I flashed and wiped due to a change in Wifi credentials and wanted to start from scratch anyways. Hardware has not changed.

Log Shows:
GYRO: No valid calibration values, please calibrate the device.
GYRO: No valid calibration values, please calibrate the device.
GYRO: No valid calibration values, please calibrate the device.
GYRO: No valid calibration values, please calibrate the device.
CFG : Configuration file does not exist.

Firmware v1.0.0 from Releases section

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on device and allow it to connect to Wifi
2. Place device horizontal
3. View web GUI. Gravity/Angle display Gyro moving
4. Click Configuration
5. Click Calibrate device
6. D1 Mini light turns solid blue
7. Device hangs and web gui is no longer responding
8. Must power cycle to return to gui

**Expected behavior**
Expect gyro to read properly and calibrate device as expected

[image]

eightheads, 2022-05-28:
I was able to re-flash once more, go through the initial setup, and calibrate at without a power cycle. Complete guess here, but I'm guessing if you don't calibrate during the initial setup, a power cycle may cause the issue I ran into. I will try and test more tomorrow with another device.

mp-se, 2022-05-28:
I just followed your steps on one of my devices and cannot really reproduce the issue. I did a clean install, configured wifi and opened up the gui. I get an angle (as expected) and also the error message that its not calibrated. 

The reason you get the text gyro moving is that the device is not able to get a valid reading from the gyro, either read error or if the values are varying too much. 

Gan you use the feature "gather support information" on the main page and post that result here, I want to check the configuration options if something there has been corrupted so it does not have the normal behaviour. It's possible to tweak the gyro read settings. 

Another thing that could be helpful is if you hook up a serial monitor and post any messages that contain the text GYRO: This output could contain some sensive data like wifi password so please remove those before posting. I would expect to find this message for instance; GYRO: Sensor is moving

mp-se, 2022-07-04:
Closing this since no new information has been provided to replicate the issue.

closed 2022-07-04 by mp-se