mp-se/gravitymon #115: Support for Lolin32 lite

daveconde, 2022-11-05:
I’ve been looking at the simplicity of putty together a floaty https://www.diyhomebrewers.com/2020/09/floaty-wifi-hydrometer-montaje-y-como.html?m=1 and I really like the simple parts list. Should gravitymon work with the lolin32 lite from this build? 

https://m.aliexpress.com/item/1005004819247383.html

mp-se, 2022-11-05:
It probably the same cpu as the esp32 d1 mini so it should work but the pins needs to be updated for it to work. I support the gyro temp as well so its not a big change, but i dont really have a device to test on. I can give it a try if you want

daveconde, 2022-11-05:
Awesome thanks, I’ll order one and see if I can get it to work

mp-se, 2022-11-05:
I'll order one as well to test the pinout configuration. From a brief look there is no way of measuring the battery level so I guess I will need to do a few more changes for this to work.

daveconde, 2022-11-05:
Thanks, I ordered all the parts earlier today and printed off the sled. I’ll hopefully have everything ready to test within a couple of weeks 😁

daveconde, 2022-11-08:
Awesome thanks, I’m just waiting on the gyro which should be here by the weekend

daveconde, 2022-11-13:
I tested it out this afternoon without the petling and it’s working great 😁 Thanks so much for adding support so quickly

mp-se, 2022-11-13:
Great, let me know if there are any issues. I will close this one now

closed 2022-11-13 by mp-se