mp-se/gravitymon #124: Push setting - HTTP 3 (GET) - doesn't send all the data

elcocowau, 2023-01-04:
When I use HTTP 3 (GET) push setting, it truncate the end of the XML data 

configure push setting HTTP 3 (GET) Ex : http://`<IP>`:`<PORT>`/`<TOKEN>`?content=
and configure format editor "HTTP option 3 GET" with this template for ex :
`
<prtg><result><channel>Densite</channel><float>1</float><value>${gravity}</value></result><result><channel>Batterie</channel><float>1</float><value>${battery}</value></result><result><channel>Temperature</channel><float>1</float><value>${temp}</value></result></prtg>
`
should return (as the test button do) : 
`
<prtg><result><channel>Densite</channel><float>1</float><value>31.7695</value></result><result><channel>Batterie</channel><float>1</float><value>4.28</value></result><result><channel>Temperature</channel><float>1</float><value>21.5</value></result></prtg>
`
But it truncate the end of the message like this : 
`
<prtg><result><channel>Densite</channel><float>1</float><value>31.7695</value></result><result><channel>Batterie</channel><float>1</float><value>4.28</value></result><result><channel>Temperature</channel><float>1</float><value>21.5</value></re
`
[image]
See wireshark capture : 
[image]

if I add dummy char at the end it truncate the dummy char and it works but it's not always "truncating" the same amount of chars.

**Software:**
 - Platform esp8266
 - Browser chrome
 - Version 1.2.0 (..90a5cb)

Goal is to send HTTP push notification to PRTG Monitoring server.

mp-se, 2023-01-17:
Sorry for the late reply but I never got a notification on this issue.

HTTP get is not intended to be used to send XML it used the format <url>?token1=value&token2=value2 according to the specification. In theory you can send an xml document as a value if you URL encode the value (which is not a feature that is implemented in gravitymon)

If you want to send XML you can do this using HTTP Post in the body of the message. So use HTTP 1 or 2. Depending on the receiving service you might need to set the right content header to tell the server that you are sending an XML document.

mp-se, 2023-01-17:
Its a little strange that the data is truncated, i will have a look what causes this. Its not strange that you get a different result on the test button since this is don in the web browser and not in the device.

elcocowau, 2023-01-18:
Hi,

First thanks a lot for sharing your great job with this firmware.

I know it's a bit strange to encode the XML in the URL but it's how the
PRTG sensor works according to the documentation

[image: image.png]

I don't find other sensor where I could send the data in the body that's
why I tried this workaround.

And yes it's even more strange that it doesn't truncate the same amount of
character. I don't find any logic.

Regards,

mp-se, 2023-01-18:
Thanks for the feedback. The playload should not be truncatated so i will treat this as a bug.

I could not spot anything strange in the code so I hope this is not a limitation in the arduino code

mp-se, 2023-01-19:
Can you try this build and see if it works for you ? https://github.com/mp-se/gravitymon/blob/dev/bin/firmware.bin

elcocowau, 2023-01-20:
Thank you for the new version of the firmware, for the moment it looks like it solved the issue.
I will try JSON format and multiple reboot asap to make sure.

mp-se, 2023-01-20:
Its just a small code change that should not affect but i will run through some tests to make sure it works

mp-se, 2023-01-25:
I've just released the official tested version, should be the same as the one that you tested.