mp-se/gravitymon #127: Brew Event

Btazbt, 2023-01-28:
It would be nice to have in the configuration an Event field.  This way data logged into a influxdb it can easily be filtered for a particular beer.  As it is now you just have to have the time range known to select the right data.

mp-se, 2023-01-29:
Here is one way to fix that. You would need to define the event and for that you can use the token fields (token1 & token2).  

When you switch brews you need to go in and change the name of the event. (token) 

Change the format template for the influx target to include the token field. Now you will have an field called event that you could filter on in influx

```measurement,host=${mdns},device=${id},temp-format=${temp-unit},
gravity-format=${gravity-unit} gravity=${gravity},corr-gravity=${corr-gravity},
angle=${angle},temp=${temp},battery=${battery},rssi=${rssi}
```

to

```measurement,host=${mdns},device=${id},temp-format=${temp-unit},
gravity-format=${gravity-unit} gravity=${gravity},corr-gravity=${corr-gravity},
angle=${angle},temp=${temp},battery=${battery},rssi=${rssi},event=${token}
```

Btazbt, 2023-01-30:
Looks like a good option. I'll give this a try soon

Btazbt, 2023-02-02:
Tested this tonight and I get exactly what I was hoping for, a quick way to filter the data for a particular event.  I'll use the token value as my brew day and then I can easily filter through the database without really needing to know dates much.

mp-se, 2023-02-04:
Updated docs, closing this

closed 2023-02-04 by mp-se