mp-se/gravitymon #150: Change wifi from config settings

Btazbt, 2023-03-17:
It would be nice to be able change the wifi connection from the configuration page too.

mp-se, 2023-03-18:
This has been raised before and its possible to do. 

But since there is not real security when reading the device configuration that would expose your wifi credentials to anyone that can access the device. That is the reason for not including them in the configuration page, now they are only exposed when accessing a private access point.

An option would be to allow the user to set the wifi password but not view it.

Btazbt, 2023-03-18:
An option would be to allow the user to set the wifi ap/password but not view it.

This seems fine. I don't think there is any reason to see the current configuration. just a list of visible wifi sources and a place to select or type in the bame to use and an entry for the password.

mp-se, 2023-03-18:
Good, I will add this to the next version. 

Add possibility to enter wifi configuration and set the password from the UI. 
Scan for current wifi networks and allow the user to select from a list. 
Current password should not be shown since there is no security on getting data from the device.

mp-se, 2023-08-05:
Added to dev branch and will be part of 1.4 release

closed 2023-08-05 by mp-se