mp-se/gravitymon #152: Documentation: BrewPi Remix service integration?

darkside90a, 2023-04-16:
Hi,
Are you interested in adding BrewPi Remix in the Service integration part of your Documentation?
BrewPi Remix v0.8 documentation describes iSpindel integration very vaguely.
I got it working after a few cut-and-tries:

----------
* In Gravitymon, take note of your Host name
(Home --> Device --> Host name:)
----------
* On your Pi running BrewPi Remix, edit this file:
/home/brewpi/settings/config.php
* add this line:
iSpindel = your-Gravitymon-host-name
* If not already set, clamp the readings to prevent wide swings:
clampSGUpper = 1.175
clampSGLower = 0.970
* After saving the changes, restart the script:
sudo systemctl restart brewpi.service
----------
* In  Gravitymon, add push settings in
Configuration --> Push settings --> HTTP 1 (POST)
* Content:
http://IP-address-of-your-BrewPi/brewpi-api.php
* Save settings and run a test push:
(Configuration --> Test push --> Test)

mp-se, 2023-04-16:
Contributions are always welcome, do you want to do a pull request yourself? Dont mind the formatting i can fix that

darkside90a, 2023-04-16:
Done! 
It was src_docs/source/services.rst

mp-se, 2023-04-17:
I cant see any open pull requests, did you start a pull request? If you need help let me know.

darkside90a, 2023-04-17:
Sorry, did changes, but forgot the last pull request part.
(Never done this before)

closed 2023-04-17 by mp-se