mp-se/gravitymon #159: Strange temperature readings

Avalonnw, 2023-08-14:
Greetings!

I was wondering if I could get some insight into this weird temperature graph I was getting:
[image]
[image]
Hardware - iSpindel on Jeffrey board.
Latest version of gravmon flashed via brewflasher.
Graphs are from Home Assistant.

Avalonnw, 2023-08-15:
1. I tried it with 12 bits, then switched back to 9. The plots are with 9
2-3. MQTT using default template. The only change was with name on the sensor.

mp-se, 2023-08-15:
Thanks, I will do a similar setup and see if I can spot any problems. I cant really see how big the difference is in your graphs. Can you check what the min / max values are in the graph so I can validate that compared to the accuracy of the sensor. 

There is a big difference in accuracy depending on what resolution is used on the temp sensor. But the increased accuracy comes with a price in conversion time = less battery life. 

9 bits = 0.5°C,  up to 93 ms
10 bits =0.25°C, up to 187 ms
11 bits = 0.125°C, up to 375 ms
12 bits = 0.0625°C, up to 850 ms

mp-se, 2023-08-15:
Ive done some tests on one of my builds. This first one is 1.5 hours with 9 bits and really stable.

[image]

This second one is over a few hours with 12 bits accuracy but there are not any real big variation that I can see. 

[image]

Could be that you have an faulty temp sensor or a copy of the dallas original.

Avalonnw, 2023-08-15:
It could be. I am not sure though. My values float within 1 degree, from 10.8 to 11.8. At the same time I have a temperature probe from ITC-308 in the wort and it's not showing any of this.
Interestingly, when I was doing formula calibration in a separate container, I haven't seen such behaviour. It has started only when I added that long metal probe for the temperature controller. Could it be some sort of interference?

mp-se, 2023-08-16:
Strange, could you take it out and run it outside your container to see if there is the same difference / fluctuation in room temperature? 

If you measured it in an active fermentation it could be real temperature since the fermentation process generates heat.

Avalonnw, 2023-08-16:
Will do, thanks.

mp-se, 2023-08-16:
This is from one of my fermentations and as you can see the temperature is fluctuating a bit. This was without cooling but even with cooling it will take time to have an effect on a batch.

[image]

Avalonnw, 2023-08-16:
It fluctuates but nothing like the sinewave I'm getting.

mp-se, 2023-08-29:
I cannot reproduce the issue, so I would believe this is a hardware related problem. 

Leaving this open for now and come back if you have more input that points to a software problem.

Avalonnw, 2023-08-30:
Thank you. I'm still getting this zigzag for some reason. Will try a
different iSpindel later.

closed 2023-10-02 by mp-se