mp-se/gravitymon #161: LOLIN D1 Mini - The Jeffrey 2.69 PCB - Extreme WiFi Unreliability

NeoMod, 2023-08-27:
**Describe the bug**
Testing the firmware on a D1 Mini v4 from LOLIN (WeMoS) mounted on "The Jeffrey 2.69" PCB highlighted what appears to be a hig unreliability with WiFi connection.

1. A "stable" WiFi connection could only be established with a dedicated access point, running an empty network with "WPA" level encryption and a simple password. Completing the Configuration proved almost impossible due to high latency and constant "Unable to get data from the device" errors. 
2. Any attempt at connecting the device to a "crowded" sensor network (WPA2 and 8-digits alfanumeric password) resulted in failure: the device would "hang" for approx 60-90 sec. and then reboot into "Hotspot" mode.
3. Further configuration after some successful attempts at connection to a dedicated WiFi network could not be completed due to random and frequent disconnections: often the device would become unreachable (even though the on-board LED still blinks) and the default index page would take 2-3 minutes to re-appear.
4. When the main page was available, readings from Temperature, Battery Voltage and Angle were correct only when using Version 1.2 of the firmware. Version 1.3 returned empty/invalid values.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash either v. 1.2 or v. 1.3 on a D1 Mini v4 assembled on a "OpenSource Distilling The Jeffrey 2.69" PCB.
5. Access the hotspot for WiFi Configuration.
6. Attempting initial configuration from the "Beer Gravity Monitor" index page or Calibration page.

**Expected behavior**
A stable wifi connection giving the ability to properly configure and test the device. 

**Screenshots**
[image]
[image]
[image]

**Software:**
 - Platform: ESP8266 LOLIN D1 MINI v4, The Jeffrey PCB 2.69
 - Browser: Chrome, Vivaldi, EDGE
 - Version: 109, 6.1.3035.302, 116.0.1938.54 

**Additional context**
The 3rd screenshot shows an error log even after what seemed to be a successful calibration attempt, with saved values been shown inside the configuration tab.

mp-se, 2023-08-28:
Thanks for the report but I dont have access to the PCB or the D1 mini v4 so I cant really reproduce the issues here. What would be helpful is if you can do the following tests to narrow down what could be the cause of this:

1. Do you have a spare D1 mini that you can try the software on without connecting any hardware ?
2. Can you try to access the GUI from a mobile device android or ios ?
3. Can you connect the device to your computer and collect the serial logs ? Dont forget to remove the battery when doing this.

mp-se, 2023-08-28:
Update! I actually found a v4 D1 mini in my stash to test with..

How do you flash the device ? I first tried using brewflasher 1.5 but that created a corrupt installation (flashing actually failed). I then tried using web.brewflasher.com and that worked. I managed to get a device up and running without issues after that.

mp-se, 2023-08-29:
On the wifi side, you can also check these suggestions when using esp8266 boards:

After looking into the specifications from espressif on the esp8266 these are some of the hardware/software limitations regarding wifi:

* Dont use space in your wifi SSID or wifi password
* Only supports 2.4Ghz bands, channels must be 1-13, other channels are not supported (In 802.11 b/g/n modes)
* Channels between 20-25 Mhz bands

From my own experiences:

* SSID must be visible (hidden SSID dont work)

Source: https://www.espressif.com/sites/default/files/esp8266_wi-fi_channel_selection_guidelines.pdf

I will add this information to the troubleshooting guide.

mp-se, 2023-10-02:
Added information to the documentation around wifi

closed 2023-10-02 by mp-se

NeoMod, 2023-11-18:
@mp-se thank you very much for all the help and the documentation. 

Sorry for my delay, got a a bit tangled with work and couldn't get back as soon as I wished. 

So, as for the first set of questions:

> * Do you have a spare D1 mini that you can try the software on without connecting any hardware ?
> * Can you try to access the GUI from a mobile device android or ios ?
> * Can you connect the device to your computer and collect the serial logs ? Dont forget to remove the battery when doing this.

1) I tried with another D1 Mini and the results where prettìy much the same.
2) I tried accessing the GUY from my Android phone and I've got mixed results: at first it seemed to be much more fluid and fast, but moving around the menu/tabs quickly caused the same issues I was noticing.
3) I haven't but I'll try asap. I tested a couple of other things and actually forgot to dump the logs. my bad "-.- I know...

> Update! I actually found a v4 D1 mini in my stash to test with..
>
> How do you flash the device ? I first tried using brewflasher 1.5 but that created a corrupt installation (flashing actually [...]

I am glad I wasn't the only one having trouble with it! I thought I was incapable of using that program: I only had one successful flash attempt, using the desktop version. The web flasher failed many times (corrupt flash) and some other times refused to download the firmware altogether. 

I cloned the GIT locally (thanks for that!) and I moved into VSCode: compiling and flashing was as simple as selecting the right COM port and pushing the button. 

(Incidentally, I have seen the same errors with BrewFlasher on another board with different ESP)

> On the wifi side, you can also check these suggestions when using esp8266 boards:

And finally I discovered pretty much the same thing! The WiFi Network itself was at fault: apparently UniFi - Ubiquiti hardware really don't play nice with ESP boards if left with default settings on gateway/AP side.

I created a dedicated network on a UNIFI UAP-AC and followed both the suggestions I found on unifi forum where there was a niche discussion about setting up gateways to allow ESP (and similar) devices to seamlessly work with standard high-usage networks in mixed environments. 

I have now applied also the suggestions you stated above, and the stability issues are gone. UI still remains a bit slow, but I'm guessing this is totally normal and expected since using and ESP8266 instead of the recommended ESP32.

On that note, I have ordered an ESP32 board and will move to it for the future tests...I definitely want to have a successful brew using a PCB loaded with gravitymon. ;)