mp-se/gravitymon #162: Unstable Wifi connection on 1.3.0 and ESP8266

misterdohl, 2023-08-28:
**Describe the bug**
When putting the GravityMon into configuration-mode, it uses everything from 5-10 minutes to connect to the network chosen. When its in "ispindel mode" It doesnt seem to be able to report every 30 mins as configured. It often can go many hours inbetween reports. 

**To Reproduce**
To be honest im unsure on how to report it, but I:
1. Use reset-pins to get the blue solid light
2. Connect to GravityMon wifi
3. Configure my wifi (using the same wifi in both slots, ive also tried to put another wifi with invalid credentials in, but no difference)
4. Press Save
5. I see that the ispindle is trying to send/connect by the blue light blinking every 30 seconds.
6. Eventually it comes online and i can reach it on a ip on the configured wifi.

When i then restart it/put it in a bowl of water it reports once (to brewfather), and then nothing. Last time i tried i had it connected for 2 hours without it reporting (every 30 min as configured).

**Expected behavior**
iSpindle should save and connect to wifi network within a minute. Should go to config mode when connecting charger and/or putting it flat on the desk.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Software:**
 - Platform: esp8266
 - Browser: Chrome
 - Version: 1.3.0 (..b68835)

**Additional context**
I can add that both my Wifi hotspot and wifi password has blank spaces in them. I dont know if this can cause some issues?
Im unable to read any logs from my router due to my provider, so i cant verify if the ispindle tries to contact dhcp etc.
Tried to output "support information", but its stuck on "collecting support data, please wait"

Output from error log:
```
HELP: Exception (28) reason=2 epc1=0x402167ac epc2=0x00000000 epc3=0x00000000 execvaddr=0x00000068 depc=0x00000000
WIFI: Failed to connect to wifi 1
WIFI: Failed to connect to wifi 7
HELP: Exception (28) reason=2 epc1=0x402167ac epc2=0x00000000 epc3=0x00000000 execvaddr=0x00000068 depc=0x00000000
WIFI: Failed to connect to wifi 1
WIFI: Failed to connect to wifi 7
HELP: Exception (28) reason=2 epc1=0x402167ac epc2=0x00000000 epc3=0x00000000 execvaddr=0x00000068 depc=0x00000000
WIFI: Failed to connect to wifi 1
WIFI: Failed to connect to wifi 7
WIFI: Failed to connect to wifi 7
HELP: Exception (28) reason=2 epc1=0x402167ac epc2=0x00000000 epc3=0x00000000 execvaddr=0x00000068 depc=0x00000000
WIFI: Failed to connect to wifi 1
WIFI: Failed to connect to wifi 7
HELP: Exception (28) reason=2 epc1=0x402167ac epc2=0x00000000 epc3=0x00000000 execvaddr=0x00000068 depc=0x00000000
WIFI: Failed to connect to wifi 1
WIFI: Failed to connect to wifi 7
HELP: Exception (28) reason=2 epc1=0x402167ac epc2=0x00000000 epc3=0x00000000 execvaddr=0x00000068 depc=0x00000000
WIFI: Failed to connect to wifi 1
WIFI: Failed to connect to wifi 7
HELP: Exception (28) reason=2 epc1=0x402167ac epc2=0x00000000 epc3=0x00000000 execvaddr=0x00000068 depc=0x00000000
WIFI: Failed to connect to wifi 7
HELP: Exception (28) reason=2 epc1=0x402167ac epc2=0x00000000 epc3=0x00000000 execvaddr=0x00000068 depc=0x00000000
WIFI: Failed to connect to wifi 7
```

Output from backup-file (removed sensitive fields):
```
{
  "meta": {
    "version": "1.3.0",
    "software": "GravityMon"
  },
  "config": {
    "mdns": "[SG]_iSpindel_Red",
    "id": "4a22c2",
    "ota-url": "https://www.gravitymon.com/firmware/",
    "wifi-ssid": "MyWifi 24Gz",
    "wifi-pass": "My Wifi Password With Spaces 1234",
    "wifi-ssid2": "OTHER_Gimmick_WIFI",
    "wifi-pass2": "OtherWifiPasswordWithNoSpaces",
    "ble": "",
    "temp-format": "C",
    "token": "",
    "token2": "",
    "http-push": "http://log.brewfather.net/ispindel?id=adsghhsdvascsdacas",
    "http-push-h1": "Content-Type: application/json",
    "http-push-h2": "",
    "http-push2": "",
    "http-push2-h1": "Content-Type: application/json",
    "http-push2-h2": "",
    "http-push3": "",
    "influxdb2-push": "",
    "influxdb2-org": "",
    "influxdb2-bucket": "",
    "influxdb2-auth": "",
    "mqtt-push": "",
    "mqtt-port": 1883,
    "mqtt-user": "",
    "mqtt-pass": "",
    "sleep-interval": 1800,
    "voltage-factor": 1.66,
    "voltage-config": 4.16,
    "gravity-formula": "0.00000095*tilt^3+-0.00013169*tilt^2+0.00737913*tilt+0.89243332",
    "gravity-format": "G",
    "temp-adjustment-value": -1,
    "gravity-temp-adjustment": true,
    "gyro-temp": false,
    "storage-sleep": false,
    "gyro-calibration-data": {
      "ax": -248,
      "ay": 497,
      "az": 1528,
      "gx": -16,
      "gy": 91,
      "gz": -11
    },
    "formula-calculation-data": {
      "a1": 21.5,
      "a2": 26.45,
      "a3": 31.07,
      "a4": 40.52,
      "a5": 49.19,
      "a6": 59.57,
      "a7": 63.32,
      "a8": 67.29,
      "a9": 71.31,
      "a10": 72.86,
      "g1": 1,
      "g2": 1.012,
      "g3": 1.024,
      "g4": 1.038,
      "g5": 1.05,
      "g6": 1.066,
      "g7": 1.072,
      "g8": 1.082,
      "g9": 1.094,
      "g10": 1.098
    },
    "app-ver": "1.3.0",
    "app-build": "..b68835",
    "angle": 89.962,
    "gravity": 1.1824,
    "battery": 4.21,
    "runtime-average": 22.78,
    "platform": "esp8266"
  },
  "advanced": {
    "gyro-read-count": 50,
    "gyro-moving-threashold": 500,
    "formula-max-deviation": 3,
    "wifi-portal-timeout": 120,
    "wifi-connect-timeout": 20,
    "push-timeout": 10,
    "formula-calibration-temp": 20,
    "int-http1": 0,
    "int-http2": 0,
    "int-http3": 0,
    "int-influx": 0,
    "int-mqtt": 0,
    "tempsensor-resolution": 9,
    "ignore-low-angles": false,
    "battery-saving": true,
    "id": "4a22c2"
  }
}
```

mp-se, 2023-08-28:
From the logs it looks like it fails to connect to the wifi and then crashes due to failed connection. 

I would suspect that there is something with you network setup that is incompatible with the esp8266, have you enabled 2.4Ghz ? Is the SSID broadcasting ?

Error 1 = No SSID found
Error 7 = Disconnected (failed to connect)

Can you hook it up to a serial monitor and get a dump of the serial log so I can see what is going on ? Just unplug the battery when you are connecting to the serial port of the esp.

misterdohl, 2023-08-29:
Looks like this is looping throughout:

```
        72 I: SDBG: Serial logging started at 115200.
        74 I: Main: Started setup for 4a22c2.
        74 I: Build options: 1.3.0 (..b68835) LOGLEVEL 4 ESP8266 
        89 I: CFG : Filesystem mounted.
        93 I: WIFI: Current reset counter 0.
       106 I: HELP: Last reset cause 28
       115 E: HELP: Exception (28) reason=2 epc1=0x402167ac epc2=0x00000000 epc3=0x00000000 execvaddr=0x00000068 depc=0x00000000
       237 I: CFG : Size of configuration file=1207 bytes.
       253 I: CFG : Configuration file /gravitymon.json loaded.
       257 W: CFG : Configuration file does not exist /hardware.json.
      1154 I: Main: Battery 5.15 V, Gyro=89.26, Run-mode=1.
fpm close 3 
mode : sta(c4:5b:be:4a:22:c2)
add if0
      1259 I: WIFI: Connecting to wifi (0) using stored settings MyWifi 24Gz.
.........................scandone
state: 0 -> 2 (b0)
.state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
cnt 

connected with MyWifi 24Gz, channel 11
dhcp client start...
..........................................................................................pm open,type:2 0
.................................................................     22612 E: WIFI: Failed to connect to wifi 7
     22707 I: WIFI: Stop double reset detection.
state: 5 -> 0 (0)
rm 0
pm close 7

     22871 W: WIFI: Failed to connect to first SSID MyWifi 24Gz.
     22872 W: WIFI: Failed to connect to any SSID.
     22907 I: TSEN: Found 1 temperature sensor(s). Using 9 bit
     22908 I: Main: Setup completed.
Fatal exception 28(LoadProhibitedCause):
epc1=0x402167ac, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000068, depc=0x00000000

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x402167ac epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000068 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3fff1d30 end: 3fff1f40 offset: 0190
3fff1ec0:  3fff09f8 3fff0a80 3fff0a00 3fff09a8  
3fff1ed0:  3fff0adc 3fff0a6c 3fff0a80 4020cc20  
3fff1ee0:  60000000 405650d1 00000001 3fff1f9c  
3fff1ef0:  3fffdad0 3fff0adc 3fff0a34 3fff1f9c  
3fff1f00:  3fffdad0 3fff09d8 3fff0a34 40216952  
3fff1f10:  feefeffe feefeffe 00000001 4020cf0b  
3fff1f20:  3fffdad0 00000000 3fff1f70 4022e410  
3fff1f30:  feefeffe feefeffe 3fffdab0 401003a1  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v000bb820
~ld
Serial connection established
```

mp-se, 2023-08-29:
After looking into the specifications from espressif on the esp8266 these are some of the hardware/software limitations regarding wifi:

* Dont use space in your wifi SSID or wifi password
* Only supports 2.4Ghz bands, channels must be 1-13, other channels are not supported (In 802.11 b/g/n modes)
* Channels between 20-25 Mhz bands

From my own experiences:

* SSID must be visible (hidden SSID dont work)

Source: https://www.espressif.com/sites/default/files/esp8266_wi-fi_channel_selection_guidelines.pdf

I will add this information to the troubleshooting guide.

misterdohl, 2023-08-29:
I reconfigured my 2.4Ghz to not have any spaces in either SSID or password, and it connected right away.

closed 2023-08-29 by misterdohl