mp-se/gravitymon #173: Bierbot as target

cparis57, 2023-12-26:
Hi All
Your custom firmware is really v very useful for its many functions. Congrats for that ! However my tech skills do not allow me to find the right parameters to transmit data to Bierbot (as a ispindel) . Is it possible to add the feature as Brewfather, Fermentrack... Or to give the configurations in the parameters of GravityMon to succeed in integrating  ispindel/Gravitymon with  Brick Bierbot. 
Many thanks in advance 
Christophe

BernhardSchlegel, 2023-12-27:
Hey there, creator of BierBot Bricks and the iSpindel Bricks implementation here. Happy to support this from the sideline.

Cheers,
Bernhard.

mp-se, 2023-12-27:
Since I have not used bierbot I dont really know how that works but it should not be a problem to send data, it all depends on the data format.

From the documentation link you shared its not really clear for me what the expected data format is. It shows an HTTP GET which is supported by gravitymon but in the ispindel code an HTTP POST (with JSON document) it done, but this is not documented in the link you shared. I would prefer to use HTTP POST since this is easier to handle and dont have any real restrictions on payload size. 

Perhaps @BernhardSchlegel can help to give me some examples on what JSON payload is accepted in the POST endpoint ?

From what I can see the bierbot service will return an interval for when the next transmission is expected, currently there is no option to parse that in gravitymon so this needs to be configured manually in gravitymon.

mp-se, 2023-12-27:
This format template seams to work:

```
{"apikey":"${token}","type":"ispindel","brand":"wemos_d1_mini","version":"${app-ver}","chipid":"${id}","s_number_wort_0":${gravity-plato},"s_number_temp_0":${temp-c},"s_number_voltage_0":${battery},"s_number_wifi_0":${rssi},"s_number_tilt_0":${tilt}}
```

Configure HTTP POST 1 to: 
``https://brewbricks.com/api/iot/v1``

TOKEN to: 
``{your apikey}``

For the HTTP POST Target add the following header:
``User-Agent: iSpindel``

cparis57, 2023-12-27:
Thanks for the guidance, I'll give it a try. I'd already suggested to  @BernhardSchlegel  the idea of using this type of parameter and here's his response: 
"Issue is, that bricks backend requires devices to obey the time until the next request. You could hot fix this by just setting a 15min interval, but that is only a workaround. A proper implementation would be better."

mp-se, 2023-12-27:
Let me know if this works. Setting the interval to 910s should secure that there is at least 15 min between requests so you dont miss any data.

I will not do any code changes to handle specific services since that will require too much testing work. The whole idea is that it should be easy to add support for new services

cparis57, 2023-12-27:
Test done as follow : 
[image]
[image]
[image]
 I recieve this message when I use the "test push button"
[image]

cparis57, 2023-12-27:
**I follow your advise :->** 
**just for information: I use http:// and not http**_s_**://**

[image]
[image]
[image]

**When I use "Testing push Target" I recieve this feedback** 
[image]

[image]

**unfortuatly  no additional spindel in Bierbot** :(

mp-se, 2023-12-27:
First, you should have added the new header to the empty field, not change the existing one. The missing one is `Content-Type: application/json`

You need to use https since http is not supported by the service (that is what error 301 means). So you cant test it using the web UI you need to do the test in gravity mode, like you have it in the wort. This is due to memory limitations on the esp8266. Testing https only works on esp32 boards.

cparis57, 2023-12-27:
what do you mean by " in the wort" ?

mp-se, 2023-12-27:
https://en.m.wikipedia.org/wiki/Wort

cparis57, 2023-12-27:
I try with your last modification, no connection to Bierbot :( maybe Berdhard will help us)

mp-se, 2023-12-28:
Thats strange, the format template worked for me but I was testing on an ESP32 not the ESP8266. Here is what I ended up with in bierbot.

[image]

This little python script will simulate the data sent so try that and see if there is some networking issue on your side. You will need python installed with the requests add-on, this can be installed via

`pip install requests`

```
import requests

host = "https://brewbricks.com/api/iot/v1"
apikey = "<your key>"

headers = {
    "Content-Type": "application/json",
    "User-Agent": "iSpindel",
}

data = {
    "apikey": apikey,
    "type":"ispindel",
    "brand":"wemos_d1_mini",
    "version":"1.4.0",
    "chipid":"AABBCC",
    "s_number_wort_0":2.345,
    "s_number_temp_0":34.56,
    "s_number_voltage_0":4.34,
    "s_number_wifi_0":-54,
    "s_number_tilt_0":33.456
}

ret = requests.post( url=host, json=data, headers=headers )
print(ret.status_code, ret.text)
```

You can change the chipid if you dont want to wait 15 min....

mp-se, 2023-12-28:
I have tested this on esp8266 and that worked fine. You can check the error log (found on the index page) and see if you have any errors reported there like this. You can set the sleep interval to like 30s so you dont need to wait 15 min for the test... 

PUSH: HTTP post failed response=500 http1

cparis57, 2023-12-28:
I did the test with your recommended parameters but still no ispindel in bierbot. Here are the error messages in the log:

GYRO: No valid calibration values, please calibrate the device.
GYRO: No valid calibration values, please calibrate the device.
CFG : Configuration file does not exist.
CFG : Configuration file does not exist.

cparis57, 2023-12-28:
> I have tested this on esp8266 and that worked fine. You can check the error log (found on the index page) and see if you have any errors reported there like this. You can set the sleep interval to like 30s so you dont need to wait 15 min for the test...
> 
> PUSH: HTTP post failed response=500 http1

I made the test witout success

mp-se, 2023-12-28:
Have you done the gyro calibration (config page) and added a formula ? You can add the default formula for testing purposes. 

Can you validate that you have gravity, angle and temperature from the startup page?

BernhardSchlegel, 2023-12-30:
Only one request: Can we please set "type" and "brand" (hardware) to correct values? 

i..e 

```
    "type":"gravitymon","brand":"the_jeffrey_2.6.9"
```

This will avoid complications in the future.

@mp-se and @cparis57 : I'd love to create a wiki page under https://docs.bierbot.com/hardware/hardware for setting up with gravitymon. Can I steal your text above and maybe get a review from you? 

Thanks,
Bernhard.

mp-se, 2023-12-30:
@BernhardSchlegel no problem to change the type to gravitymon, I will update the docs with that,. 

Since GM does not have any specific hardware and runs on the same hardware as iSpindel the hardware value needs to be set by the user based on what they have. So what options do you want them to consider ? GM also supports the Floaty hardware so that needs to be considered as well. Besides the standard esp8266 I also support a wide range of ESP32 variants, is this something to consider in the brand part? I can add an option that sets the CPU board automatically like esp8266, esp32, esp32 (floaty), esp32s2, esp32s3, esp32c3 etc.

closed 2024-02-23 by mp-se