mp-se/gravitymon #175: CraftBeerPi 4

DindonMaitre, 2024-01-20:
Hello, I would like to configure my iSpindle to send information to CraftBeerPi 4. I saw that currently there are only instructions for BrewBlox. Could you provide similar instructions for CraftBeerPi 4? Thank you very much.

mp-se, 2024-01-20:
If you have any documentarion on how to make it work i can add that to the docs, i am not familiar with that brewing system. 

But it should not be a problem as long as it can handle http or mqtt requests

DindonMaitre, 2024-01-21:
Here's the Github : https://github.com/PiBrewing/cbpi4-iSpindle , it explains installation and settings . Sorry I'm not an expert

mp-se, 2024-01-21:
Based on those instruction the post settings should be quite easy. You only need to add the URL under one of the POST options. Make sure there are no other warnings in the UI for things that needs to be defined. 

http://[your-cbpi4-ip]:8000/api/hydrometer/v1/data

DindonMaitre, 2024-01-21:
It works thanks , just when i test , i have this : Starting test push target 'http-2'
Push target 'http-2' failed with error code 500

mp-se, 2024-01-21:
500 means internal error so the server crashed, you can try to change the format template to ispindel standard or see if you get any errors in the server log21 jan.

mp-se, 2024-05-27:
Closing this since there are no activity

closed 2024-05-27 by mp-se