mp-se/gravitymon #187: New iSpindel build. Config Unable to Calibrate gyro C3 Beta 2.0.0

Hwerow, 2024-08-22:
**Describe the bug**
A clear and concise description of what the bug is.
Unable  to configure the gyro click the button says configured Home page says not
also Temperature sensor not recognised but works ok in 1.4.1

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...' Configuration
2. Click on '....' initialise gyro save page
3. .says configured but on the home page says not

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Software:**
 - Platform (esp8266, esp32) esp32-C3
 - Browser [e.g. chrome, safari] Firefox
 - Version [e.g. 22] 1229.0.1 64 bit

**Additional context**
Add any other context about the problem here.

mp-se, 2024-08-22:
I'm aware of this issue on the c3 board. From what I have been able to figure out it's an issue with arduino / esp-idf framework so I havent been able to find a way around this yet. Upgrading to arduino 3.0 is something i'm looking into to see if that can fix the problem. So for now the 1.4 version is the latest one that works with the c3 board. I have focused on getting the software working and have not dug into the details with each board.

mp-se, 2024-08-23:
Same as #184

mp-se, 2024-08-27:
Finally found the issue so this will be fixed included in the next release. The drawback is that it was the LED that was in conflict so I have to disable that functionality on the ESP32c3

mp-se, 2024-08-27:
Once the build is completed you can try the version in the dev branch under bin, that should now work on the c3

closed 2024-08-27 by mp-se

Hwerow, 2024-08-27:
Thanks Magnus. Small correction Home menu IP Adress should be Address. Cheers Peter