mp-se/gravitymon #191: Sending to targets triggers exception

mburgess00, 2024-09-04:
**Describe the bug**
Exception happens after movement detected. MQTT for Home Assistant and HTTP for Brewfather configured.

**To Reproduce**
Steps to reproduce the behavior:
1. Have gravitymon on
2. Move gravitymon to trigger movement detected

**Expected behavior**
Send data to targets

**Screenshots**
Here is the serial trace while this happens:
 (500)	129	1757	5254.
                                          70260 I: GYRO: Movement detected (500)	3747	2674	3409.
                                                                                                                  70575 I: WEB : webServer callback for /api/ping.
                                    71270 I: GYRO: Movement detected (500)	542	817	190.
                                                                                                         72787 I: Main: Sending data to all defined push targets.
                                   72789 I: PUSH: Free-heap 25 kb, Heap-rag 10 %, Max-block 22 kb Stack=320 b.
                                                                                                                   72809 I: PUSH: Template loaded from disk /http-1.tpl.
                                          72810 I: TPL : Buffer needed 255.
                                                                                72811 I: TPL : Transformed template 290 chars to 255 chars
            72813 I: PUSH: Sending values to HTTP post
                                                           72817 I: PUSH: Adding header 'Content-Type': 'application/json'
                                                                                                                               73601 I: PUSH: HTTP post successful, response=200
                                                  73613 I: PUSH: Template loaded from disk /mqtt.tpl.
                                                                                                          73616 I: TPL : Buffer needed 1342.
              73623 I: TPL : Transformed template 1336 chars to 1342 chars
                                                                               73623 I: PUSH: Sending values to mqtt.

                                                                                                                     --------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

Exception (4):
epc1=0x401004d8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

**Software:**
 - Platform esp8266
 - Browser firefox
 - Version 2.0.0-beta2

**Additional context**
Manual test of targets works.

mp-se, 2024-09-05:
Do you have more of the text after Soft WDT reset so I can pinpoint the exact location of the crash?

mburgess00, 2024-09-05:
Here's the whole stack trace:

                              --------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

Exception (4):
epc1=0x401060f7 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffff3b0 end: 3fffffd0 offset: 0160
3ffff510:  00000000 0004ad8d 7ba5e353 0568ab96  
3ffff520:  3ffff570 4021439c 00000034 0004ad8d  
3ffff530:  00001388 3fff6224 3ffffae0 40214492  
3ffff540:  3fff449c 3ffe8368 40237944 40237934  
3ffff550:  fffffffc 3fff08d4 3fff08c0 3ffff600  
3ffff560:  0000075b 4021439c 3ffffae0 40213b28  
3ffff570:  40238760 08bc10ac 40238760 08bc10ac  
3ffff580:  3fff241c 3fff22d4 3ffff600 402377a6  
3ffff590:  3fff22bc 3fff22d4 00000000 40230a10  
3ffff5a0:  00000000 0004ad8c 2353f7ce 3fff22a4  
3ffff5b0:  0000000d 4023ba1c 00000020 40101124  
3ffff5c0:  3ffffa7c 3ffff67c 0000000d 4022f4ea  
3ffff5d0:  3fff2194 3fff22a4 3ffe8368 4023ba1c  
3ffff5e0:  3fff22a4 0000075b 3ffff600 00000001  
3ffff5f0:  3ffffa7c 0000075b 3ffffae0 4022353b  
3ffff600:  00000200 00000200 3fff5da4 3fff78bc  
3ffff610:  0001000a 00002710 00000000 3ffffae0  
3ffff620:  3fff241c 40238760 00000000 0000075b  
3ffff630:  00000000 00000000 00000000 00000000  
3ffff640:  40237494 3ffff67c 00000000 3fff08c0  
3ffff650:  00000001 3ffff67c 00000000 4022b314  
3ffff660:  3ffffae0 0004ad8c 00000000 00000000  
3ffff670:  0004ad8c 00000000 00000000 00000001  
3ffff680:  00000000 00000000 00000200 00000200  
3ffff690:  3fff78bc 3fff5da4 402304f4 3ffff634  
3ffff6a0:  3ffff660 40230660 40237760 3ffff664  
3ffff6b0:  3ffff670 40230634 402305fc 00000000  
3ffff6c0:  00000000 0000e600 00000000 00000000  
3ffff6d0:  00000000 00000000 00000025 40212c6f  
3ffff6e0:  3ffff740 3ffff720 00000018 402be6d9  
3ffff6f0:  40237494 3ffe8cfb 3fff08c0 3ffffa68  
3ffff700:  3fff22bc 3ffffbe0 3fff22d4 40101124  
3ffff710:  3fff6e1c 00000540 3ffffa7c 4022c96e  
3ffff720:  00000000 00000536 00000536 3fff22bc  
3ffff730:  0000075b 3fff22a4 3ffffae0 40223718  
3ffff740:  3ffffa7c 0000053e 3ffffa7c 4022ca03  
3ffff750:  3ffffa7c 0000053e 3ffffa7c 3ffffa68  
3ffff760:  3ffffa7c 3ffffae0 3ffff780 4020e227  
3ffff770:  43990000 4080d75f 3ffeec00 40201250  
3ffff780:  646d7b24 007d736e 07000000 3fff2404  
3ffff790:  000b000f 80000000 64697b24 0000007d  
3ffff7a0:  05000000 38663036 40006366 06000000  
3ffff7b0:  6f747b24 7d6e656b 08000000 00000000  
3ffff7c0:  7f97b01a 00000000 6f747b24 326e656b  
3ffff7d0:  0900007d 00000000 00000000 00000000  
3ffff7e0:  65747b24 007d706d 07000000 352e3732  
3ffff7f0:  00000030 05000000 65747b24 632d706d  
3ffff800:  0900007d 352e3732 00000030 05000000  
3ffff810:  65747b24 662d706d 0900007d 352e3138  
3ffff820:  3fff0030 05000000 3fff2434 000c000f  
3ffff830:  80000000 00000043 00003000 01000000  
3ffff840:  61627b24 72657474 0a007d79 33302e34  
3ffff850:  3ffff900 04000000 3fff4474 0011001f  
3ffff860:  80000000 00303033 4023b049 03000000  
3ffff870:  3fff3f44 0012001f 80000000 00003038  
3ffff880:  4023f8b4 02000000 3fff4424 000b000f  
3ffff890:  80000000 2e363033 3f003030 06000000  
[... 95 lines cut ...]
3ffffe90:  00000000 3fff0dfd 00000002 4022ee09  
3ffffea0:  00000001 00000068 00000000 3fff0ddc  
3ffffeb0:  00000000 0004a948 374bc6a7 0563bb7f  
3ffffec0:  00000000 00000002 00000000 3fff0ddc  
3ffffed0:  00000002 00000002 00000002 40228da6  
3ffffee0:  000003e8 0004a948 00000068 00000041  
3ffffef0:  3fff0858 3fff089e 3fff0850 4022822c  
3fffff00:  40105fe5 05b10f3e 3fcb851f 3fff0896  
3fffff10:  40106092 3ffeb024 05b10f3e 00000000  
3fffff20:  40106269 05b10f3e 3fff1114 00000000  
3fffff30:  3fff03e0 3fff1114 3ffe865c 3fff1114  
3fffff40:  3fff0900 0004a9c3 96872b02 05644a26  
3fffff50:  0004a9c2 3fff1114 00000001 00000001  
3fffff60:  42b07775 00000000 00000000 40340000  
3fffff70:  00000000 403b8000 a0000000 40560eee  
3fffff80:  00000000 40044210 3fff0878 3fff113c  
3fffff90:  3fffdad0 00000000 3fff08b8 4020d333  
3fffffa0:  3fffdad0 00000000 3fff08a8 4020d3b3  
3fffffb0:  00000000 00000000 3fff1110 4022e0dc  
3fffffc0:  feefeffe feefeffe 3fffdab0 40100459  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

I did figure out this was with the "home assistant 2" profile, with the "home assistant" profile I don't see this.

mp-se, 2024-09-05:
Are you using SSL with the MQTT connection ?

mburgess00, 2024-09-05:
I don't think I had SSL enabled as I saw that an 8266 wouldn't have enough memory to handle that. Either way I'm fine with just the home assistant profile.

mp-se, 2024-09-08:
It looks like the watchdog comes from not properly handling a webserver request so its not connected to the push function, but I have not found why this happends.

closed 2024-10-07 by mp-se