mp-se/gravitymon #193: No temperature sensor detected when upgrading to 2.0.0-beta3

mburgess00, 2024-09-09:
**Describe the bug**
The UI reports no temperature sensor detected after upgrading from 2.0.0-beta2 to 2.0.0-beta3

**To Reproduce**
Steps to reproduce the behavior:
1. flash 2.0.0-beta3
2. observe the error

**Expected behavior**
temperature sensor should work

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Software:**
 - Platform esp8266
 - Browser firefox
 - Version 2.0.0-beta3

**Additional context**
Opensourcedistilling Jeffrey 2.69

mp-se, 2024-09-09:
Can you try the build in the dev branch? Ive changed the temp library which seams to fix this on some of my builds. I have seen this on ocassion myswlf

mburgess00, 2024-09-09:
It does seem to work on the dev branch build

mp-se, 2024-09-10:
Perfect, that confirms my suspicion.  It fixed the issue on some of my build. I will make this the standard version for the next build. Wonder if that could have something to do with clone sensors ?

mp-se, 2024-09-11:
Does it work with the ispindle software?

phariseo, 2024-09-11:
@mp-se - I do have an original Dallas DS18B20 (100% original - from the authorized distributor), and it behaves the same on Beta3.
worked flawlessly on older 1.x branch of GravityMon, works flawlessly on Dev as well.

mp-se, 2024-09-11:
> @mp-se - I do have an original Dallas DS18B20 (100% original - from the authorized distributor), and it behaves the same on Beta3. worked flawlessly on older 1.x branch of GravityMon, works flawlessly on Dev as well. Board is an Jeffrey 2.69

This is really strange, my code is the same and has not changed in quite a while. Must be some combination of libraries that I use that does not work in the same way... not really easy to pinpoint. 

The OneWire library has been used since the beginning but its not really maintained any more, but the new OneWireNG seams to work better. I actually changed some temp sensors on my builds before I figured out it was a software problem

mburgess00, 2024-09-11:
Okay did some testing on one unit. iSpindel 7.3.3 firmware does work

iSpindel-7.3.3:
[image]

Gravitymon 2.0.0-beta2
[image]

Gravitymon 2.0.0 dev branch
[image]

mp-se, 2024-09-12:
Great news, then that is confirmed. I will use the new library in the future.

closed 2024-09-30 by mp-se