mp-se/gravitymon #195: Discussion about adding new ICM support and FIFO support

Levi--G, 2024-09-30:
Hi,

So i wanted to contact you about features i'd like to contribute and maybe discuss this in advance so its clear on how you might want these added. I have quite alot of experience with these IMU's over the years and since i started making my ispindels noticed some improvements that could be made.

First of all i'd like to add support for the ICM-42670-P chip, this is another recent invensense IMU that is very low power and has great accuracy and has some nice features that could really augment my readings while saving power. To do this i suggest I create an interface to abstract the IMU, and change the IMU field to be a pointer assigned on detection. If this is fine with you of course. If you want me to do it in some other way i'm all ears.

For the FIFO i'm surprised it isn't used yet. It would leave the IMU on and collecting data while the main IC sleeps and then when the ESP wakes up all the samples are already there, collected over the entire time it was sleeping thus being more accurate over time and shortening the time the esp is online. I've already ran some tests on this and it seems very promising for this scenario. With MPU6050 i get a very consistent 0.01° accuracy using FIFO. On the MPU6050 however it might use more power than sleeping (i will measure exactly how much soon). But maybe adding it as an alternative option could be beneficial? I can fork/change the code and add the option but i don't know how you make your front-end and might need some help on that side. But first i will do the power/accuracy measurement on both the new and old imu's so you can see the difference.

Regarding the accuracy/power consumption comparison i will do, do i just test the default settings (every 15m push), or is there other data you are interested in to know if you want to add it? Basically how longer the sleep time, the more fifo will use compared to all sleeping, but the larger the accuracy difference as well.

If you don't like these features and rather not have them pulled into the main, i can always maintain my own fork, thats fine for me too. ;)

Thank you for the already amazing project and hope to hear from you soon.
Levi

Levi--G, 2024-10-01:
Ok will do :)

You are right, using FIFO would give the temp sensor an offset, but not more than you already have now, either way is the temp sensor of the MPU not really that good, i see a few degrees offset even with accel and gyro disabled. The gyro is bad for battery life but it gets duty cycled in fifo mode, so i will test how much impact it really has very soon. Ill use the dev branch and keep you updated.

mp-se, 2024-10-01:
On the gyro temp sensor, i did quite a lot of tests before adding that feature and I agree that the accuracy is not that good, but comparable to having a DS18 on low resolution and you read it as soon as the chip started. After a while it will continue to increase and be way off and it requires at least 10 minutes to cool down again....

But do your own tests and we can compare notes later.

mp-se, 2024-10-12:
From what I could find out reading the data sheet a MPU6050 (only chip) would draw 3.5 mA which would run on an 18650 battery for 38 days. But to extend that one could change the the sleep behaviour and allow the FIFO to work in intervals.  For instance of there i a sleep interval of 15 minutes

1. Deep sleep (esp + mpu) for 10 minutes
2. Wakeup and enable FIFO for MPU, let esp go back into deep sleep
3. Wakeup and read mpu, push data and then repeat. 

This would extend the battery life quite a lot but still allow for better readings if I understand the fifo function.

Levi--G, 2024-10-13:
So the MPU uses 3.5mA while measuring and enabled, when using fifo you actually normally slow down the measurements and put the gyro in a mode that will only enable it while taking a fifo sample, this should lower the power usage alot. The higher the data rate the more it uses over time. I will try to get some accurate current measurements soon about this.

Theoretically you can slow down the measurements of the mpu to +-4Hz. Meaning only 4 times/s the gyro will wake and be used. With a 4kB memory and 14B per reading that means you can store 285 samples or 71 seconds maximum without data loss, however the fifo replaces old data when full so you just always have the last minute of data at your disposal when waking after that time. This is not alot, and its one of the reasons im going to add ICM-42670-P support. 

For the ICM: 
The gyro on the ICM uses much less power (0.55mA for gyro+accel in continuous use) and it can slow down to virtually any data rate, meaning the fifo will be adjustably filled according to the deep sleep time so power/storage can be optimally used. the MPU doesn't have those features yet (and has been discontinued for many years now). I have made a GY-521 compatible breakout for the ICM and its on the way from the PCB manufacturer, i will share results as soon as it arrives. Of course the design will be shared with the community once tested and if people request them i can also sell them premade.

My next actions:
Since i believe FIFO will only really shine on the ICM i will initially be comparing MPU in normal mode to ICM in fifo mode on the esp8266 (classic spindle board). Once those results are in and fifo seems worth it i can backport it to the mpu and find out if we need to tweak the power consumption to stay in line with the "classic" mode.

mp-se, 2024-10-13:
I agree that the ICM options is much more interesting to look into, seams like a much better chip.  Where to you buy the chips from? I would want to get a few so I can create a test setup. 

One thing I lack with the standard iSpindle hardware is the missing interrupt pin connection, that could help to reduce the power consumption since one could start the measurement and then let the esp do the networking stuff. Having to manually check for status consumes a lot of unnecessary battery (i think the read takes almost 800 ms) which is quite a lot of wasted time.

Levi--G, 2024-10-13:
> I agree that the ICM options is much more interesting to look into, seams like a much better chip. Where to you buy the chips from? I would want to get a few so I can create a test setup.

The chips you can buy from any IC seller, Mouser is really good and has large stocks, but if you want to save some bucks you can buy them from china as well, however make sure to check the whoami register on receiving them, they can be fakes or different chips. Make sure you have the needed tools to solder these tiny smd chips though, its harder that it looks. 😉  Since i do this alot for me its no issue. If everything goes well and the PCB/tests work i have no issue sending you a sample board.

> One thing I lack with the standard iSpindle hardware is the missing interrupt pin connection, that could help to reduce the power consumption since one could start the measurement and then let the [...]

Yes i have also seen the same, atm the I2c readout without any processing is taking 930ms. I have also seen the setup also takes a fair chunk of this time which can be improved without changing any functionality. But ill come back to that in a later stage. The interrupt won't really improve that i believe since the time between the readings will still be the same. You could use the fifo at a high rate and then read them at once maybe before putting the chip to sleep? But you would still need to check the fifo byte count i guess. FYI just the fifo reading takes 0.56ms/sample so +-28ms for 50samples.

I did some more testing and apparently i need to rectify smth, i said earlier the gyro consumption would be less due to the cycling, apparently the mpu6050 doesn't support that yet so it will continuously use 3.82mA sadly. So i don't think continuous fifo will ever be viable for the mpu without a loss in battery lifetime. 😞

Levi--G, 2024-10-13:
So for temporary reference measurements i will be using (on average):
ESP8266 + MPU8266: 7.339s using 67.07mA (492.26mC) on average before going back to sleep @ 170µA for 900s (153mC)
In total that is 645.26mC per cycle of +-15min
If you have different measurements feel free to share them 😉 

So next week when my boards arrive i can do a proper comparison with those values.

mp-se, 2024-10-13:
It was a long time since i did any measurements but i use this to calculate battery life on an esp8266: 15 mA / day (sleep) and 160 mA per hour when active (this include temp and gyro). My last fermentation had an average runtime of 4.2 seconds

Levi--G, 2024-10-19:
Okay so the boards arrived, are soldered and are working. So good so far. I'm now adding multi IMU support to the code, however while doing so i noticed you are not using default VScode/c++ formatting, could you please share your settings so i can format the code in the same way? :) I'd hate to mess up your formatting.

mp-se, 2024-10-19:
I use precommit for formatting, there is a config in the root ditectory.

Levi--G, 2024-10-19:
Sadly i could not get the precommit to work, I'll sort that out later.

On the bright side i got classic angle measurements working today, so i can do some 1on1 comparison with the mpu tomorrow. The ICM seems to have way less jitter and inaccuracies so if the current really is lower even without using the FIFO i see that as a win/upgrade already ;)

Sadly i did make a mistake on the pcb and missed a mandatory resistor i now soldered externally. So most likely ill have to make a new pcb first before others can use it.

mp-se, 2024-10-20:
It wont work on windows, you need linux or macos. I use wsl on my windows machine

Great that you are making progress, looking forward to the results

Levi--G, 2024-10-20:
Ah that explains alot 😆 

Okay so the bare minimum power consumptions are in, to make these tests as fair as possible i eliminated all delays the firmware introduces due to wifi connection and reading json etc by just loading an empty sketch with the exact code from the imu handling. The ICM needs a longer setup delay (45ms) to smooth out the gyro according to the datasheet, so the setup time is longer than the MPU, i think the MPU might technically also need this, but since that is never done i left the code as is. So even though the ICM uses the extra delay and is more stable, due to the higher data rate the reading is shorter and it overall uses less power. So overall even without using different techniques like the fifo the ICM42670-p uses only 19.1% of the power the MPU uses. I am planning on improving this more by optimising some calls, and afterwards ill try to compare to the fifo method with the ICM.

PS: due to the lower power consumption the temperature also has a smaller offset 👍 😉 

IMU | Sleep | Setup Max | Setup Avg | Setup Time | Full Max | Full Avg | Full Time | Time per sample | Total Charge | Charge per sample
--- | --- | --- | --- |--- |--- |--- |--- |---|---|---
MPU6050 | 30µA | 10.52 mA | 5.26 mA | 2.77 ms | 11.43 mA | 5 mA | 930.8 ms | 18.616 ms/S | 4.65 mC | 93 µC/S
ICM42670-p | 39µA | 4.92 mA | 556 µA | 45 ms | 4.92 mA | 1.05 mA | 85 ms | 1.7 ms/S | 89 µC | 1.8 µC/S

mp-se, 2024-10-20:
Good work, that is positive for the power consumption even though is quite small compared to the consumption of the esp. But in combination with ble transmission is should have an bigger effect on battery life

would be interresting to see the accuracy as well, how much that could be improved.

Levi--G, 2024-10-20:
It is indeed small compared to the esp, but the shorter readout (10x shorter) will also be seen in the total consumption since the esp can go sleep sooner, which is the next measurements i will do. Plus this might make it a candidate to use the fifo which could make for much better accuracy over time.

Regarding accuracy i have no way of measuring that in numbers, although the ICM sensor is more accurate by default, unless i spend a week pulling statistics i have no way of comparing them. What i might do is put an mpu and an icm in the same fermenter next time to compare the results? But that will not be for a few weeks.

mp-se, 2024-10-20:
Yes, the shorter read time will probably have a bigger impact on the total consumption. Wifi is till the biggest battery killer according to my previous testing. 

Would be interresting to get comparable values, from the same device to compare them. It should be possible to have both sensors attached to a device and send data from, which then could be compared

Levi--G, 2024-10-20:
Yes my current setup is without an actual soldered ISpindel, just an ESP8266 (the same one with the same firmware) where i swap the IMU just by plugging a new one in 😉 

In that setup i now get the following results with the adjusted firmware for multi IMU support (that doesn't change anything to the MPU in theory) and without an external temp sensor:

|             | MPU     | unit | ICM      | unit |
| ----------- | ------- | ---- | -------- | ---- |
| Sleep avg   | 197     | µA   | 187      | µA   |
| Sleep time  | 900     | s    | 900      | s    |
| Sleep       | 177,3   | mC   | 168,3    | mC   |
| Active avg  | 65      | mA   | 69       | mA   |
| Active time | 4,434   | s    | 3,375    | s    |
| Active      | 288,21  | mC   | 232,875  | mC   |
| cycle time  | 904,434 | s    | 903,375  | s    |
| cycle       | 465,51  | mC   | 401,175  | mC   |
| %           | 100     | %    | 0,861797 | %    |

As you can see the +- 1s time gain means about 14% of energy gain per cycle.

PS: i don't know why but once attached to the ESP the MPU sleep usage is now higher, i think it was originally leaching off the I2C voltage somehow which didn't get measured, now i measured the entire circuit so the only power source was my measurement device. This should be more accurate. So even in sleep the ICM seems to use less now which corresponds to the datasheets.

Levi--G, 2024-10-20:
Preliminary results for fifo mode with ICM suggest it uses 1/5th the power of the mpu in fifo mode. but that would still be twice as much overall compared to a normal cycle. However since there is no setup (the gyro is already on) i need to verify how much is recuperated due to the shorter esp cycle. But imo this comes close to actually being possible to use. In this mode there is a (very accurate) sample stored in fifo every 10s, compared to 50 samples (in 1s) every 15mins this should give an accuracy that is worlds apart.

tldr; ICM in fifo uses 665µA and ill test more next week after i added FIFO mode in the gravitymon code.

mp-se, 2024-10-20:
Sounds promising, i managed to find a breakout board on aliexpress that i ordered so i can run the code. Should be here in 2 weeks

Levi--G, 2024-10-20:
I don't know where you live but i can send one of mine if you don't mind the external resistor/shipping 😉 

The aliexpress board isn't ispindel layout compatible btw, so don't forget to switch the pins around. (clk should be the second pin for ispindel layout)

mp-se, 2024-10-20:
I live in sweden, you can email me on info@gravitymon.com and we can continue this discussion there

mp-se, 2024-10-20:
> I don't know where you live but i can send one of mine if you don't mind the external resistor/shipping 😉
> 
> The aliexpress board isn't ispindel layout compatible btw, so don't forget to switch the pins around. (clk should be the third pin from VCC for ispindel layout)

I saw it had a different pinout but that s not a problem, will wire it to my dev board

mp-se, 2024-10-21:
You can email me on mambos.trim-0j@icloud.com instead. I forgot i removed the mailservice for that domain.

Levi--G, 2024-10-27:
Sorry i didn't get back sooner, I was busy with implementing and further testing FIFO with the ICM.

Regarding temperature: The ICM with its lower power consumption seems to have less issues with heating in FIFO mode, i tested the temperature and compared it to room temp and it consistently stays within <2°C from room temp, now i might just be lucky with my chip, but i like to think that isn't the case here. The stability is also reversed to the MPU, temp becomes stable the longer it has been since the setup/startup, since that phase uses more power.

Regarding power consumption:
| Sleep avg   | 750,97  | µA |
| ----------- | ------- | -- |
| Sleep time  | 900     | s  |
| Sleep       | 675,873 | mC |
| Active avg  | 71      | mA |
| Active time | 3,21    | s  |
| Active      | 227,91  | mC |
| cycle time  | 903,21  | s  |
| cycle       | 903,783 | mC |
| %           | 194,149 | %  |

The relative usage is sitting at +-200% so we are looking at half battery life compared to normal (measured on ESP8266, this will be better on ESP32). This is however close enough to consider adding at least as a compiler option for people who know what they are doing 😉 

I'm also looking at the even newer ICM chips that can improve even further on this, but they are sitting at a 4€ per chip, which seems a bit much just to save some battery...

Either way except for some cleaning up i believe my current code has all the main features, how do you want to proceed on this? I expect to have a real life test this week if everything goes right and ill throw in some different boards (MPU, ICM and ICM fifo) to compare accuracy.

Levi--G, 2024-10-27:
The current FIFO mode works as follows:

- ESP configures IMU at power cycle according to settings (this needs work since changing sleep time setting will not work correctly atm)
- IMU gathers 138 samples or as close to it as possible during the sleep time
- ESP wakes and checks if the IMU is already configured => if yes skips setup phase
- ESP reads all samples
  - All samples are individually checked for movement and discarded if needed
  - All good samples are averaged and converted
- Angle and temp are used as normal
- ESP sleeps again without sleeping the IMU

So far it seems very stable and accurate but the remaining issue is after changing the sleeptime the imu needs to be reset, and that is not easily possible atm, any ideas?

mp-se, 2024-10-28:
Great news, I'm still waiting for my chip from china so I cannot do any testing/running the code until i receive that so please do some accuracy testing as well that would be very interesting to see. 

On the sleep cycle, there is some logic in the code that changes the sleep cycle, especially if there is an error with WIFI connection, the device will go into sleep for 60 seconds and then retry again.

I think i will need to dig into the code and interface for the IMU to get a better understanding on how that works. 

Feel free to open a PR when you think your code is ready. I will let you know when my board has arrived.

Levi--G, 2024-11-11:
Sorry ive been busy the last week.

Currently i would like to address a few issues first before I'd consider a PR:
- More testing (obviously)
- Fix my board. Currently my board with the ICM chip on it started acting weird and doesn't wake from deep sleep every so often, i don't know if this is a hardware issue or a software issue so this needs investigating (but i'm fairly sure its hardware)

FiFo only issues (a PR for just supporting the new IMU would not need these fixed):
- Mitigation for the gui/constant measure mode, currently this results in only one value every few tried readings due to the different data gathering
=> could be fixed if we use a different api/method for gui reads or for publishing reads, this way the IMU code can detect which read to use for the IMU
- Mitigation for the issue with changing time in fifo mode requiring a power cycle to reset the IMU
=> could be fixed if we have a callback from the webserver or configuration every time it changed or is saved so we can reconfigure the IMU
- Investigate the current code flow when no valid samples are returned, i think this isn't working properly with fifo atm, angle seems to just default to 0 instead of (not) registering a bad read
- Either add FiFo mode option to gui or decide which will be default or available? 
=> Adding the option will need a feedback to the imu similar to the time change callback, this could be a shared callback to just reload the imu

I can implement these solutions but they will need time to work out. I have some leave coming up so i should be able to work on them soon 😄

mp-se, 2024-11-17:
You can always configure the IMU based on the startup mode, so in gravity it used the sleep timeout and when in config mode a constant read mode is used ? Then you should not need to do a power cycle if sleep timeout is changed. 

Error handling is key to detect when there is no valid data or some of the logic will be way off... 

I finally got my sample board from china last week so when you do a PR I can also do some testing (or at least run the code).

Levi--G, 2025-01-22:
Sorry it took so long for me to respond,

I checked the hardware issue i was having and noticed all the esp8266 i bought for this project have a flash issue, this causes the esp to never boot from deep sleep. I ordered a replacement batch and they have the same flash chip and thus have the same issue. So im now forced to search a way to get working esp devices again or solder a new flash ic on them, either way i can't work on this until i have new replacements sadly...

I'll let you know once i have a working board again 😉

Levi--G, 2025-01-26:
Some things i noticed in the last days picking up the ICM again:
- My initial findings of saving power by skipping imu setup is really saving power on ESP32 based boards since the cycle is much shorter and esp32 uses much more power while cpu is active. This means the fifo might actually save power compared to the traditional cycle.
- As it stands an ESP32S2+ICM uses less battery than a traditional ESP8266+MPU in gravity mode 😄 
- The accuracy is much, much better, the annoying jumps in mpu graphs are completely gone with the icm, i will share some results in a few days
- FIFO mode seems to not need external filtering at all anymore, which i always needed with mpu based gravmons and even when in motion for a part of the 5mins sleep time it keeps a very steady result

During this testing invensense came with another imu that provides even lower power gyroscope and more fifo space, i don't think i need it since the 42670 one is already a world of difference, but if someone would be interested in it, I am willing to add software support for it. Plus it seems my breakout would be 1-1 compatible, so it can serve for that as well.

mp-se, 2025-01-27:
Really good news, looking forward to do some tests on this. Do you have any plans to do a complete pcb for this build?

Levi--G, 2025-01-28:
> Really nice, if you are considering a complete pcb then the kegland container could be an good option.

You mean the rapt pill one? Tbh i don't really like that one as the seal is always in liquid and its the hardest part to clean, i was actually just considering the classic ispindel pet bottle, but with wireless charging, reed switch and fully hot glue sealed. 😄 

> I'm working on some large refactoring of the codebase so I can support other type of sensors as well, currently I'm focused on creating a variant with pressure sensors to measure either fermentation pressure or keg pressure (you can see the refactoring efforts in the pressuremon branch). I will not do any big changes to the gyro part but would appreciate if you see any changes to other parts of the codebase or the UI to handle the new gyro.

I have seen your effort and im really excited to see the progress but on the other hand im still using the old main and merging will be absolute hell... I am fully prepared of supporting any new imus i add to stay compatible once merged, granted that the changes don't make it near impossible 😉 

> One thing would be to show in the UI what gyro is detected and used. I dont know if there should be any configuration options connected to the new gyro available in the UI ?
> 
> I guess most of the current settings would not be required such as sample rate etc.

Sample rate and some other settings won't be used in fifo mode, maybe adding a fifo mode toggle would be all that is needed as i implemented both on the IMU just in case. All other settings get converted automatically to the respective settings needed for the imu.

mp-se, 2025-01-28:
the new branch (pressuremon) is up to date with the dev branch (fully merged) so unless there are big changes it should not be a big deal to merge. Many changes are splitting the codebase and renaming files. 

Good to know that there will not be that many changes, i guess the gyro type can be easily detected using the iic adress. 

I have some ispindle pcb lying around so I will build a rig that I can install and test with once you are happy with the changes.

Levi--G, 2025-01-28:
> the new branch (pressuremon) is up to date with the dev branch (fully merged) so unless there are big changes it should not be a big deal to merge. Many changes are splitting the codebase and [...]

Thats what i was trying to say, i did make a lot of changes and had already implemented some multi imu support which meant a lot of changes in the same files you changed a lot in afterwards on the dev branch 😄 I'm about 110 commits behind at the moment. My branch is based on the 2.1.0-beta1 from back in october 😉 So it will take some time to merge and prep a PR.

On the upside the new imu has been working stable for days now so i think once i get the source to merge we should be good to go.

mp-se, 2025-01-29:
Great, just base it on the dev branch that is up to date with the latest changes. Looking forward to testing this.

Levi--G, 2025-02-01:
This is come data to compare the ICM (in blue) to the MPU (in Yellow).
This is the tilt data to not have the calibration interfere, and i didn't really compare actual accuracy, but its clear that the ICM in fifo mode has no spikes unlike the MPU that even with no active fermentation has these weird spikes and inaccuracies. The other thing that i noticed is that even in very active fermentation the ICM has a smoother line. And even in movement of adding new liquid its much more stable and representative of the actual situation.

In terms of temperature data the ICM has a lower resolution sadly, but i think the data is really close and with a moving average filter you could still get a nice graph if you wanted. But i think most people looking for accurate temperatures will use a better (external) sensor either way.

[image]

mp-se, 2025-02-01:
This look really promising, looking forward to try this out

mp-se, 2025-04-13:
When do you plan to create the PR ? Just wondering if I should wait and include this in the 2.2 release or not

Levi--G, 2025-04-13:
> When do you plan to create the PR ? Just wondering if I should wait and include this in the 2.2 release or not

Sorry, but with all the changes you did i cannot simply merge it and i need to write everything i changed again while trying to figure out what you changed. I have started on it before but due to the large amount of changed i could not finish it before a new load of changes happened. Maybe i can find a weekend in the following weeks to do it in one go but i won't be able to properly test it for quite a while sadly.

mp-se, 2025-04-13:
Push a merge to the 211 branch and i can fix the rest, im probably quicker in doing so

Levi--G, 2025-04-13:
> Push a merge to the 211 branch and i can fix the rest, im probably quicker in doing so

The current code is still on 2.1 beta 😄 
I'll work on it next week

Levi--G, 2025-04-14:
@mp-se Am is supposed to merge to Dev or 211? Dev is both ahead and behind 211?

mp-se, 2025-04-14:
Merge to 211 that should be closest to what you started with, i can always adjust later

mp-se, 2025-04-15:
211 only contains a fix for the lite (floaty) version so that should not be a conflict.

mp-se, 2025-04-19:
Hi! I had a look at your code and I would suggest that the initialization of the gyro is changed. In your dev branch both gyros are created and only one is used. This would consume more RAM which is already in short supply on the esp8266 boards so it might be that the SSL support will not work due to lack of RAM.

I would suggest that you do a i2c scan and check which gyro exist (by i2c adress) and then just create the instances that is required for that hardware setup.

Levi--G, 2025-04-19:
Okay so i don't mind dynamically instantiating the gyro like you suggest but keep in mind/consideration:
- Microcontrollers don't like dynamic memory allocation which will be really slow compared to the current setup, making startup time take longer and thus battery usage higher.
- The actual memory used is not that much, its just a few bytes + the variables in the class which would end up at maybe a saving of 20-30 bytes (ill get an exact number in a bit).
- The json system uses much more, so if you really want to keep supporting esp8266 it would be much better to take a look at that, since it allocates a lot of memory and is very slow on almost all microcontrollers.
- The esp8266 is basically dead, the ESP32c3 that more recently came out is my new goto controller for low power consumption, it uses less power than esp8266 while being faster and having better ram/cpu. I will even start replacing my existing controllers especially since espressif have abandoned the sdk updates for esp8266 which now remains buggy and in the future potentially unsafe (no wpa3 support, vulnerabilities won't be fixed, etc).

The board i plan on designing next will be made for the ESP-c3 supermini combined with one of the new imus. I am currently experimenting with the c3 since it has 8kB RTC memory that stays on during deep sleep, possibly skipping the entire setup phase and allowing more advanced scenarios such as history, filters, .... And among others save what imu is used and what address it is at ;)

Levi--G, 2025-04-19:
For info i tested it and allocations are:
MPU6050 mem usage: 44 bytes
ICM42670p mem usage: 52 bytes
so dynamically allocating will save max 52 bytes in best scenario, is that worth having to split up detection code from running code and somehow having to pass detection variables to a dynamically allocated class? (Im not conviced it is.) Im working on merging, you can always change it later if you want.

mp-se, 2025-04-19:
> For info i tested it and allocations are: MPU6050 mem usage: 44 bytes ICM42670p mem usage: 52 bytes so dynamically allocating will save max 52 bytes in best scenario, is that worth having to split [...]

Well static allocation typically goes to the stack and dynamic to the heap. The stack is the main issue on the esp8266 so even 50 bytes can make a difference since the stack is really small. If 8266 is not to be supported then it does not really matter. 

I know that Json is memory heavy but since that and ssl is not used at the same time then it will not cause any issues. SSL requires around 21kb memory to be used and on the 8266 we are really close to that. A payload larger than 1200 bytes will cause a crash.

mp-se, 2025-04-24:
Lets continue the discussion here since the PR is closed. I liked your idea of using FIFO mode when doing measurements and non FIFO when in configuration mode. 

Perhaps we can use the chip reset reason to determine what the mode of operation will be ? If its a wakeup from deep sleep is should be measuring and then the mode should be FIFO but if its other reasons we could start the gyro in the other mode? What to you think ?

The new structure should be flexible enough to add many more gyro options in future releases. I will try to remove any dependencies towards other classes in project so that the gyro development can be handled separately thus making it faster to try out changes without the full codebase. 

Ps. I have added the flash write flag as a separate feature request that I will fix separately, thanks for the suggestion.  Ds.

Levi--G, 2025-04-24:
> Lets continue the discussion here since the PR is closed. I liked your idea of using FIFO mode when doing measurements and non FIFO when in configuration mode.
>
> Perhaps we can use the chip reset [...]

How you determine the mode of operation does not really matter for the gyro, so ill leave that to you. 😄 All that matters is that it gets passed to the gyro on setup so the gyro can then be setup according to the running mode. We will also need to call this when the json/mode gets updated to make the changes have effect immediately since some options get set in the setup (since fifo mode does not always actually setup since it needs to keep running to work and config resets the imu). maybe along with a force flag to force setup upon config change?
Something like: 
`virtual bool Setup(config mode, bool force) = 0;`
where config is an enum of the different modes?

> The new structure should be flexible enough to add many more gyro options in future releases. I will try to remove any dependencies towards other classes in project so that the gyro development can be handled separately thus making it faster to try out changes without the full codebase.

That is great, im looking forward to it, since debugging the gyro was a real issue when the gravmon kept going to sleep 😄
It will also keep all gyro related code contained as an added bonus meaning all that ever would be needed to support a new imu would be the implementation of the interface, making my job real easy 😄  

> Ps. I have added the flash write flag as a separate feature request that I will fix separately, thanks for the suggestion. Ds.

Thank you so much 🙏 

PS: what branch are you working on right now? because i can't seem to locate your new code? Then i can take a look at the interface and maybe write a mockup?

mp-se, 2025-04-24:
> How you determine the mode of operation does not really matter for the gyro, so ill leave that to you. 😄 All that matters is that it gets passed to the gyro on setup so the gyro can then be setup [...]

It would be great if I can get your help to set this up since you have written the code for now, at least change it so its possible to switch between the modes. It would be good if the switch can be made as automatic as possible so that users dont need to care about that fact. But lets test a few things and see what works. 

I also added a flag to switch from X to Y axis so I can use the boards i bought without modifcations.

> That is great, im looking forward to it, since debugging the gyro was a real issue when the gravmon kept going to sleep 😄 It will also keep all gyro related code contained as an added bonus meaning [...]

If you connect the PIN_CFG1 and PIN_CFG2 then it will not go to sleep, a useful option for debugging the gyro stuff...

> Thank you so much 🙏
>
> PS: what branch are you working on right now? because i can't seem to locate your new code? Then i can take a look at the interface and maybe write a mockup?

I havent checked in any code yet, but it will be merged into the dev branch when I'm done, i'm testing compiling all the targets at the moment so I will do a check-in soon. The UI code has been updated to match the changes I have done so far. There are still some stuff to be handled and these are the things that pre-commit has noted in the code;

src/ICM42670P_gyro.cpp:35:  Using C-style cast.  Use static_cast<int16_t>(...) instead  [readability/casting] [4]
src/ICM42670P_gyro.cpp:37:  Using C-style cast.  Use static_cast<uint16_t>(...) instead  [readability/casting] [4]
src/ICM42670P_gyro.cpp:250:  Using C-style cast.  Use static_cast<size_t>(...) instead  [readability/casting] [4]
src/ICM42670P_gyro.cpp:280:  Using C-style cast.  Use static_cast<uint16_t>(...) instead  [readability/casting] [4]
src/ICM42670P_gyro.cpp:316:  Using C-style cast.  Use static_cast<int16_t>(...) instead  [readability/casting] [4]
src/ICM42670P_gyro.cpp:317:  Using C-style cast.  Use static_cast<int16_t>(...) instead  [readability/casting] [4]
src/ICM42670P_gyro.cpp:318:  Using C-style cast.  Use static_cast<int16_t>(...) instead  [readability/casting] [4]
src/ICM42670P_gyro.cpp:319:  Using C-style cast.  Use static_cast<int16_t>(...) instead  [readability/casting] [4]
src/ICM42670P_gyro.cpp:320:  Using C-style cast.  Use static_cast<int16_t>(...) instead  [readability/casting] [4]
src/ICM42670P_gyro.cpp:321:  Using C-style cast.  Use static_cast<int16_t>(...) instead  [readability/casting] [4]
src/ICM42670P_gyro.cpp:322:  Using C-style cast.  Use static_cast<int16_t>(...) instead  [readability/casting] [4]
src/ICM42670P_gyro.cpp:280:  Add #include <algorithm> for min  [build/include_what_you_use] [4]
Done processing src/ICM42670P_gyro.cpp

mp-se, 2025-04-24:
I can setup a separate project with the gyro code and a mockup for testing and development tomorrow and add you as a collaborator to make it easier to work on the gyro code. I have a similar setup for my pressure sensor option

Levi--G, 2025-04-25:
Btw what Clang format are you using? I just cannot get my brackets on same line since i use MSVC formatting by default.

mp-se, 2025-04-25:
> Btw what Clang format are you using? I just cannot get my brackets on same line since i use MSVC formatting by default.

I use the one defined in the pre-commit default setting, should be this one; https://github.com/bmorcos/pre-commit-hooks-cpp

I just use the default settings and pre-commit handles the formatting. If you have windows you can install and run pre-commit in a wsl instance. But dont put too much effort into that one, I can run pre-commit when merging the code

Levi--G, 2025-04-25:
Since this was about the 42670-p which got merged, you can also close this if you want, unless you want to keep it open until next release?

mp-se, 2025-04-26:
Lets keep it open until its been merged documented and tested

mp-se, 2025-04-26:
Would be good if you can help me with some text dor the docs around how the gyro works and the different modes, what benefits you see from your testing. You can attach it here since i havent restructured the docs yet

mp-se, 2025-04-26:
@Levi--G  can you also fix the pre-commit warning/errors in the code ?

Total errors found: 5
src/ICM42670P_gyro.cpp:33:  Using C-style cast.  Use static_cast<int16_t>(...) instead  [readability/casting] [4]
src/ICM42670P_gyro.cpp:35:  Using C-style cast.  Use static_cast<uint16_t>(...) instead  [readability/casting] [4]
src/ICM42670P_gyro.cpp:256:  Using C-style cast.  Use static_cast<size_t>(...) instead  [readability/casting] [4]
src/ICM42670P_gyro.cpp:286:  Using C-style cast.  Use static_cast<uint16_t>(...) instead  [readability/casting] [4]
src/ICM42670P_gyro.cpp:326:  Using C-style cast.  Use static_cast<int16_t>(...) instead  [readability/casting] [4]
src/ICM42670P_gyro.cpp:327:  Using C-style cast.  Use static_cast<int16_t>(...) instead  [readability/casting] [4]
src/ICM42670P_gyro.cpp:328:  Using C-style cast.  Use static_cast<int16_t>(...) instead  [readability/casting] [4]
src/ICM42670P_gyro.cpp:329:  Using C-style cast.  Use static_cast<int16_t>(...) instead  [readability/casting] [4]
src/ICM42670P_gyro.cpp:330:  Using C-style cast.  Use static_cast<int16_t>(...) instead  [readability/casting] [4]
src/ICM42670P_gyro.cpp:331:  Using C-style cast.  Use static_cast<int16_t>(...) instead  [readability/casting] [4]
src/ICM42670P_gyro.cpp:332:  Using C-style cast.  Use static_cast<int16_t>(...) instead  [readability/casting] [4]
src/ICM42670P_gyro.cpp:286:  Add #include <algorithm> for min  [build/include_what_you_use] [4]

Levi--G, 2025-04-28:
> Would be good if you can help me with some text dor the docs around how the gyro works and the different modes, what benefits you see from your testing. You can attach it here since i havent restructured the docs yet

Sure, which part do you need me to do/add? The only 1 to 1 comparison i still have the data from was posted here: https://github.com/mp-se/gravitymon/issues/195#issuecomment-2628976977
The main benefit is better accuracy, less jitter and less missed/retried data due to movement in furious fermentation 😄 Once i get my new board going though it might just be lower power as well, if i switch to SPI the readout of data should be 24x faster and thus wake time much shorter, combined with the fast resume i added last week that should mean the esp should be awake much shorter.

Levi--G, 2025-04-28:
> [@Levi--G](https://github.com/Levi--G) can you also fix the pre-commit warning/errors in the code ?
> 
> Total errors found: 5 src/ICM42670P_gyro.cpp:33: Using C-style cast. Use static_cast<int16_t>(...) instead [readability/casting] [4] src/ICM42670P_gyro.cpp:35: Using C-style cast. Use static_cast<uint16_t>(...) instead [readability/casting] [4] src/ICM42670P_gyro.cpp:256: Using C-style cast. Use static_cast<size_t>(...) instead [readability/casting] [4] src/ICM42670P_gyro.cpp:286: Using C-style cast. Use static_cast<uint16_t>(...) instead [readability/casting] [4] src/ICM42670P_gyro.cpp:326: Using C-style cast. Use static_cast<int16_t>(...) instead [readability/casting] [4] src/ICM42670P_gyro.cpp:327: Using C-style cast. Use static_cast<int16_t>(...) instead [readability/casting] [4] src/ICM42670P_gyro.cpp:328: Using C-style cast. Use static_cast<int16_t>(...) instead [readability/casting] [4] src/ICM42670P_gyro.cpp:329: Using C-style cast. Use static_cast<int16_t>(...) instead [readability/casting] [4] src/ICM42670P_gyro.cpp:330: Using C-style cast. Use static_cast<int16_t>(...) instead [readability/casting] [4] src/ICM42670P_gyro.cpp:331: Using C-style cast. Use static_cast<int16_t>(...) instead [readability/casting] [4] src/ICM42670P_gyro.cpp:332: Using C-style cast. Use static_cast<int16_t>(...) instead [readability/casting] [4] src/ICM42670P_gyro.cpp:286: Add #include for min [build/include_what_you_use] [4]

This should be ok now

[3 more replies]

closed 2025-04-29 by mp-se