mp-se/gravitymon #196: ESP8266 LED conflict

Levi--G, 2024-10-13:
**Describe the bug**
Hi just creating a separate issue i encountered while converting a Ispindel to gravitymon for testing to keep the other chat clear. So im running stock GravityMon on stock ISpindel hardware with a MPU6050 attached. But im not getting a connection to the gyro on the current dev channel. After analysing with logic analyser i saw the LED pattern on the SCL line. I checked the code and for ESP8266 the LED_BUILTIN pin is currently equal to SCL, meaning once the led is used I2C will no longer work.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash stock firmware on esp32
2. Attach logic analyser to pin 2 (SCL)

**Expected behavior**
LED usage should not interfere with I2C

**Screenshots**
[image]

**Software:**
 - Platform: esp8266
 - Browser: any
 - Version: dev branch

**Additional context**
Is it possible to disable the led in config or defines so i can confirm? i didn't see the option but i could have missed it.

mp-se, 2024-10-13:
You are correct, its the same pin. I had the same issue with the C3 board but there the gyro didnt work at all.. 

The code for the led is in the espframework and there is no define to disable it. Simplest way is to comment out calls to ledOn and ledOff.

mp-se, 2024-10-13:
The strange thing is that I have not noticed any problems, i have done a few brews without issues. thanks for spotting this. i will do a release later today.

Levi--G, 2024-10-13:
No problem and thank you for the quick responses and the great project 😄

closed 2024-10-13 by Levi--G

mp-se, 2024-10-13:
I noticed that my espfwk also used the led functions in some scenarios so I ended up adding a define to disable the LED interaction in the framework instead, just to be sure. The dev branch is now updated

commit bc9e1d4a (2024-10-13): Fix #196 Solve ESP8266 LED conflict
commit f966e7c2 (2024-10-13): Merge pull request #197 from Levi--G/dev

Fix #196 Solve ESP8266 LED conflict