mp-se/gravitymon #198: 2.0.0 -> 2.0.1 firmware update fails 8266

joshuascheid, 2024-10-14:
**Describe the bug**
Using the firmware.htm interface to upload the firmware.bin image from 2.0.1 results in a solid LED and no HTML site.  Attaching to the serial interface shows continuous "garbage" output.  Power cycle and reset application do not appear to help.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash 8266 iSpindel to 2.0.0 using brewflasher
2. Use firmware.htm interface to upload 2.0.1 firmware.bin
3. Observe solid LED

**Expected behavior**
Successful upgrade to 2.0.1

**Screenshots**
```
                                                                    b$�scl

                                                                          cd�;sd

                                                                               cl�{���#�

        쎜�
           #���s�

                 d`��n$�s��nogc�l#l`rd

                                      �

[10:11:45.369] Disconnected
```

**Software:**
 - Platform esp8266
 - Browser firefox on Linux
 - Version 2.0.1

**Additional context**
None

mp-se, 2024-10-14:
You probably have a corrupted flash,, try to reinstall 2.0 firmware and do a full erase via brewflasher and then try the upgrade again.

I have uploaded the update to 3 of my own spindels without issue.

joshuascheid, 2024-10-14:
Upgrade success after full erase.

closed 2024-10-14 by joshuascheid