mp-se/gravitymon #202: Old iSpindel configuration cannot be deleted.

makuip, 2024-12-16:
Gravmon reports that an iSpindel configuration was found and offers to import or delete it. When I click on "delete" an error message "Page not found! /other/links is not a valid URL for this application!" appears.

Steps to reproduce the behavior:
1. Take a device with an old iSpindel configuration.
2. Click on 'delete' in the notification that the iSpindel configuration can be imported or deleted.

I would expect to see a page where I can delete the old iSpindel configuration.

Software:
 - Platform: esp8266, Version, 2.0.1
 - Browser: Edge, Version 131.0.2903.86

[image]

mp-se, 2024-12-16:
thanks, I will fix that.

mp-se, 2024-12-18:
Fixed in beta 2

closed 2024-12-18 by mp-se