mp-se/gravitymon #204: Unable to set 0°P in the gravity formula.

Punt27, 2024-12-18:
**Describe the bug**
If you are in the °P mode it's not possible to set 0°P as the first point. And you can't set an °P over 10. I calibrated the device in SG and switched to °P but I haven't tried on an Sud yet. So not sure if the calibration is working.

**Expected behavior**
Can you set the software to automatically change the value when you switch form SG to °P and back? Ore make it work with the necessary values for°P. From 0 up to 20

**Screenshots**

**Software:**
 - Platform (esp32)
 - Browser [chrome phone & PC]
 - Version 2.0

mp-se, 2024-12-19:
You can try the build in the dev branch that should fix the problem so you can enter 0 for plato.

mp-se, 2024-12-20:
Let me know how you testing works out. I haven't really spent time on the Plato support since i'm not using that myself but any feedback on improvement is appreciated.

mp-se, 2025-01-13:
Should be fixed in 1.1

closed 2025-01-13 by mp-se

Punt27, 2025-05-11:
Sorry for the late response. It seems to have worked. I haven't been able to do any brewing for a while, but as far as I can see, it should have worked. Thanks for the implementation.