mp-se/gravitymon #209: Wiring Schematics for ESP32C3 not correct

Haggenti, 2025-01-16:
according to file main.hpp :

#define PIN_SDA 7
#define PIN_SCL 6
...
#define PIN_DS A0
#define PIN_VOLT A3

So, wired it as the following

[gravitymon-ESP32C3mini.pdf](https://github.com/user-attachments/files/18438245/gravitymon-ESP32C3mini.pdf)

And it works nicely in gravity mode, but when in deep sleep, no wake-up possible. 
What is the correct wiring for EN pin ? I was not able to my myself the pin for wake-up in the source code.

Thanks

mp-se, 2025-01-17:
You should not need the BAT43 diode on the esp32 it does not use the EN for deep sleep wakeup it has internal mechnisms for that feature. 

See example here; https://randomnerdtutorials.com/esp32-deep-sleep-arduino-ide-wake-up-sources/

Haggenti, 2025-01-17:
ok.
Tried it today, but not working : it stays in deep sleep mode, until i do a manual reboot

mp-se, 2025-01-18:
Its designed to have the same pinout as a esp8266 so all boards created for the ispindel should work. The addition is the resistor for the voltage divider that needs to be added since the esp32 does not have an internal resistor as the esp8266 does. On my builds i have added the BAT42 diode but that should not matter.  

You could try to connect the diode between EN and pin 1. Here is an example with the diode, https://opensourcedistilling.com/ispindel/

I have tested the C3 but i prefer the S3/S2 boards since the WIFI in the C3 is not that great.  There are two variants of the C3 mini but that should not matter in this case.

Haggenti, 2025-01-20:
I got it working with BAT42 between EN and pin 2...

Thank you for your help !

closed 2025-01-20 by Haggenti