mp-se/gravitymon #212: Sleep interval always 60mins + serial debugging on ESP32 not working

makuip, 2025-01-27:
I upgraded one of my iSpindels from ESP8266 to ESP32s3 mini. So far everything is working fine. However, I do have the problem that the sleep interval is always 60mins, independent of the push settings. When the device is in config mode, e.g. lying on the table, values are pushed at the right intervals (e.g. 300s). It looks like that the device is not waking up from sleep properly.

To provide some logs, I wanted to do some debugging while monitoring the serial console. However, compiling the software with the option DUSE_SERIAL_PINS failed. This is the error message:

```
Compiling .pio/build/gravity32s3-release/libcd0/espframework/serialws.cpp.o
.pio/libdeps/gravity32s3-release/espframework/src/log.cpp: In constructor 'SerialDebug::SerialDebug(uint32_t, bool)':
.pio/libdeps/gravity32s3-release/espframework/src/log.cpp:84:50: error: no matching function for call to 'HWCDC::begin(const uint32_t&, SerialConfig, int, int)'
   EspSerial.begin(serialSpeed, SERIAL_8N1, 44, 43);
                                                  ^
In file included from /home/martin/.platformio/packages/framework-arduinoespressif32/cores/esp32/HardwareSerial.h:54,
                 from /home/martin/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:196,
                 from .pio/libdeps/gravity32s3-release/espframework/src/ArduinoLog.hpp:19,
                 from .pio/libdeps/gravity32s3-release/espframework/src/log.hpp:27,
                 from .pio/libdeps/gravity32s3-release/espframework/src/log.cpp:24:
/home/martin/.platformio/packages/framework-arduinoespressif32/cores/esp32/HWCDC.h:59:10: note: candidate: 'void HWCDC::begin(long unsigned int)'
     void begin(unsigned long baud=0);
          ^~~~~
/home/martin/.platformio/packages/framework-arduinoespressif32/cores/esp32/HWCDC.h:59:10: note:   candidate expects 1 argument, 4 provided
*** [.pio/build/gravity32s3-release/libcd0/espframework/log.cpp.o] Error 1

```

**Software:**
 - Platform (esp8266, esp32): ESP32s3 mini
 - Browser [e.g. chrome, safari]: Firefox 133.0
 - Version [e.g. 22]: 2.1.0

**Additional context**
Don't know how to provide additional logs/information.

mp-se, 2025-01-27:
Check your battery level, probably battery saving mode that kicks in

makuip, 2025-01-27:
Thanks, that solved the problem.

closed 2025-01-27 by mp-se