mp-se/gravitymon #213: ESP32 d1 mini littlefs file not found, no permits for creation

BrutoCane, 2025-02-07:
I'm having some issues using GravityMon on an ESP32 D1 Mini from AZ-Delivery. I was trying to send some test data to BrewSpy, so I set up the HTTP POST, URL, and an interval of 30/60 seconds. Everything seems to be working fine with the push test, but no data is received by the service. So, I checked the log, and this is what I saw:
```24304 I: WEB : webServer callback for /api/status(get).
[ 24312][E][vfs_api.cpp:105] open(): /littlefs/gravitymon.json does not exist, no permits for creation
25439 I: WEB : webServer callback for /api/status(get).
[ 25447][E][vfs_api.cpp:105] open(): /littlefs/gravitymon.json does not exist, no permits for creation
25839 I: WEB : webServer callback for /api/ping.
28087 I: WEB : webServer callback for /api/ping.
28303 I: WEB : webServer callback for /api/status(get).
[ 28309][E][vfs_api.cpp:105] open(): /littlefs/gravitymon.json does not exist, no permits for creation
32294 I: WEB : webServer callback for /api/status(get).
[ 32300][E][vfs_api.cpp:105] open(): /littlefs/gravitymon.json does not exist, no permits for creation
32804 I: WEB : webServer callback for /api/ping.
35150 I: WEB : webServer callback for /api/ping.
35766 I: WEB : webServer callback for /api/status(get).
[ 35772][E][vfs_api.cpp:105] open(): /littlefs/gravitymon.json does not exist, no permits for creation
36288 I: WEB : webServer callback for /api/status(get).
[ 36294][E][vfs_api.cpp:105] open(): /littlefs/gravitymon.json does not exist, no permits for creation
39675 I: WEB : webServer callback for /api/calibrate.
39750 I: WEB : webServer callback for /api/ping.
40276 I: WEB : webServer callback for /api/status(get).
[ 40282][E][vfs_api.cpp:105] open(): /littlefs/gravitymon.json does not exist, no permits for creation 
.....>......-2136.00000,	389.00000,	758.00000,	26.00000,	-45.00000,	6.00000
41720 I: CFG : Configuration saved.
41722 I: CFG : Stored wifi settings in EEPROM as backup.
42137 I: WEB : webServer callback for /api/ping.
43753 I: WEB : webServer callback for /api/calibrate/status.
43790 I: WEB : webServer callback for /api/config(read).
44380 I: WEB : webServer callback for /api/status(get).
[ 44388][E][vfs_api.cpp:105] open(): /littlefs/gravitymon.json does not exist, no permits for creation
46840 I: WEB : webServer callback for /api/ping.
48371 I: WEB : webServer callback for /api/status(get).
[ 48379][E][vfs_api.cpp:105] open(): /littlefs/gravitymon.json does not exist, no permits for creation
49081 I: WEB : webServer callback for /api/ping.
51742 I: WEB : webServer callback for /api/config(write).
51946 I: CFG : Configuration saved.
51948 I: CFG : Stored wifi settings in EEPROM as backup.
52360 I: WEB : webServer callback for /api/status(get).
[ 52367][E][vfs_api.cpp:105] open(): /littlefs/gravitymon.json does not exist, no permits for creation
53793 I: WEB : webServer callback for /api/ping.
56153 I: WEB : webServer callback for /api/ping.
56355 I: WEB : webServer callback for /api/status(get).
[ 56363][E][vfs_api.cpp:105] open(): /littlefs/gravitymon.json does not exist, no permits for creation
60757 I: WEB : webServer callback for /api/status(get).
[ 60765][E][vfs_api.cpp:105] open(): /littlefs/gravitymon.json does not exist, no permits for creation
```

So, I tried manually uploading gravitymon.json from the advanced tools, and the problem disappeared. However, now I have another issue when trying to send data to brewspy:
```1115 I: Main: Battery 4.29 V, Gyro=89.99, Run-mode=1.
1215 I: WIFI: Connecting to wifi using stored settings Carlesso_wifi.

1332 I: WIFI: Connected to wifi wifi_name ip=192.168.50.148.
1333 I: WIFI: Using mDNS name gravitymon4d63f0.
1364 I: TSEN: Found 1 temperature sensor(s). Using 9 bit
1365 I: Main: Activating web server.
1366 I: WEB : Configuring web server.
1367 I: WEB : Configuring web server.
1409 I: WEB : File=error.log, 65 bytes
1426 I: WEB : File=gravitymon2.json, 2111 bytes
1443 I: WEB : File=http-1.tpl, 225 bytes
1453 I: WEB : File=reset.dat, 1 bytes
1454 I: WEB : Setting up async web handlers.
1457 I: WEB : Web server started.
[  1472][E][vfs_api.cpp:105] open(): /littlefs/runtime.log does not exist, no permits for creation
....
[182930][E][vfs_api.cpp:105] open(): /littlefs/push.dat does not exist, no permits for creation
182959 I: PUSH: Template loaded from disk /http-1.tpl.
182960 I: TPL : Buffer needed 209.
182960 I: TPL : Transformed template 225 chars to 209 chars
182971 I: PUSH: Sending values to HTTP post
182971 I: PUSH: HTTP, SSL enabled without validation.
182982 I: PUSH: Adding header 'Content-Type': 'application/json'
184565 I: PUSH: HTTP post successful, response=200
[184573][E][vfs_api.cpp:182] remove(): /push.dat does not exists or is directory
243050 I: Main: Sending data to all defined push targets.
243051 I: HELP: Heap 114 kb, FreeSketch 1792 kb.
[243056][E][vfs_api.cpp:105] open(): /littlefs/push.dat does not exist, no permits for creation
```
Occasionally, data is sent and reaches BrewSpy (I received 2 records in 20 minutes) as you can see:
[image]

I tried increasing sleep interval and push timeout but its not working

**To Reproduce**
Flash GravityMon on an ESP32 D1 Mini, check log from serial monitor

**Software:**
 - Platform esp32 d1 mini az-delivery
 - Browser edge / brave
 - Version 2.1.0

**Additional context**
I tried uploading GravityMon 2.0.0 to an ESP8266 D1 Mini clone, and it seems to be working fine, but it crashes after each deep sleep (with both GravityMon and iSpindel fw). So, I switched to an ESP32 to check whether the issue is related to software or hardware.

I will receive five more ESP8266 D1 Mini boards from AZ-Delivery (should be better then this clone im using now) tomorrow and will test them as well.

mp-se, 2025-02-07:
Those are not errors from gravitymon, its messages from the arduino framework. All messages with [E] in them can be ignored.  When files do not exist after flashing you will get those messages which is normal. Gravitymon.json is the name of the old config which which should not exist on a new install and if its there it will try to migrate to the new format.

On the interval it will probably go into battery saving since there is no valid voltage reading, check the voltage and calibrate it to be correct. 

If there are any valid error those will be in the error.log file.

mp-se, 2025-02-07:
If you have issues i would recommend to do a full erase and flash it again

BrutoCane, 2025-02-07:
> Those are not errors from gravitymon, its messages from the arduino framework. All messages with [E] in them can be ignored. When files do not exist after flashing you will get those messages which [...]

First of all, thanks for the response.
I flashed an ESP8266 D1 Mini (clone), and it didn’t show any errors, so I thought there could be a problem on the esp32.

However, I still have a problem with my ESP8266 D1 Mini. When connected via USB, it successfully pushes data to Ubidots (BrewSpy isn’t working). But as soon as I disconnect it, it pushes data only once, then fails to wake up from deep sleep and shows some memory errors (?).

[image]

This is my ESP8266 D1 Mini V3.0:
https://a.aliexpress.com/_EwoNHMs

Do you think the issue could be with this hardware?
I just bought this one, should it work?
https://www.amazon.it/dp/B08BTH77F3

Or is there something else you would recommend I order? I’m using Jeffrey PCB 2.69.

As for the ESP32, I tried a clean install, and it still shows all those errors. However, I was able to push data to Ubidots, but not to BrewSpy, so the issue is not with your software.

mp-se, 2025-02-07:
Wakeup errors on the d1 is typically hardware related,  check that the resistor or diode is correctly installed.

Im not a hardware guy but i usually buy from the lolin store on aliexpress. Those are original boards.

On the memory dump it looks like there that is a watchdog reset (wdt) and i need to dig into where in the code that happends, i guess you are using the esp8266 and latest version. Could be that there is a timeout waiting for a response from brewspy server… check the format template to see if therre is something there. Does it happen with other services?

BrutoCane, 2025-02-07:
> Wakeup errors on the d1 is typically hardware related, check that the resistor or diode is correctly installed.
>
> Im not a hardware guy but i usually buy from the lolin store on aliexpress. Those are [...]

I used a BAT43, and it is correctly installed. I wanted to buy one from the Lolin store on AliExpress, but they recently released the new V4 board, which doesn’t have enough documentation on how to install it on the iSpindel. For example, it doesn’t include the diode.

Yes, I was using an ESP8266 with the latest version. That happened as soon as I connected the device to WiFi.

Now i tried again with esp8266 and latest version, it went deep sleep and then printed this

[image]

mp-se, 2025-02-07:
That looks normal

BrutoCane, 2025-02-07:
> That looks normal, you can always try with the original ispindle sw and see if the same issues at least to rule out hardware issues

on the original firmware i have the exact same problem, i think its a hardware issue

BrutoCane, 2025-02-08:
So, i received my new esp8266 d1 mini from azdelivery, they seems to be working fine and is sending data to ubidots without any problem.
I dont understand why they cant connect to my asus rog ax6000, there are a lot of esp8266 and esp32 connected to it and those are the only one giving me headache
They can only connect to the hotspot of my smartphone

[image]

mp-se, 2025-02-08:
Check the qa in the docs for wifi settings that the esp can handle, not uncommon that they dont work with some routers

BrutoCane, 2025-02-08:
> Check the qa in the docs for wifi settings that the esp can handle, not uncommon that they dont work with some routers

I tried creating another WiFi network on my ASUS router, set a static IP for the ESP, and it connected instantly.
I really don’t understand why, since the network settings are the same as those of my primary WiFi network.
Thank you for your support.

closed 2025-02-08 by BrutoCane