mp-se/gravitymon #217: fails to detect the temperature and gravity sensors

xxyzuo, 2025-04-01:
Describe the bug
The system fails to detect the temperature and gravity sensors after flashing iSpindel. This issue occurs on both version 2.1.0 and 2.0.0. The development board used is ESP8266 D1 Mini, and the mainboard is [iSpindel-Gravitymon-Pill-PCB](https://github.com/andreq/iSpindel-Gravitymon-Pill-PCB). Since the sensors work correctly after flashing iSpindel, hardware failure can be ruled out.

Expected behavior
The firmware should recognize and correctly read data from both the temperature and gravity sensors.

Software:
Platform: ESP8266 D1 Mini
Firmware Version: 2.1.0, 2.0.0
Mainboard: [iSpindel-Gravitymon-Pill-PCB](https://github.com/andreq/iSpindel-Gravitymon-Pill-PCB)

mp-se, 2025-04-01:
Its not likely that this is a software error, I'm using an esp8266 for my main sensor and this has worked for the last 6 months without issues. I would guess a bad clone or the esp or sensor.

xxyzuo, 2025-04-04:
Thanks for the reply, I performed the same operation on all four of my devices and the results are the same, so this shouldn't be a hardware problem!  
I flashed firmware version 2.0.0 via brewfather, 2.1.0 flashed with an error message, so 2.1.0 flashed firmware.bin from the upgrade on gravitymon's web.

mp-se, 2025-04-04:
Does any of the older versions work?

xxyzuo, 2025-04-05:
The original ispindle firmware 7.3.3 works fine. And I tested 2.0.0 beta1-beta3, and 2.1beta3 beta1, doesn't work!

mp-se, 2025-04-05:
Have you tried 1.4? Just trying to see if any version works

mp-se, 2025-04-05:
Can you post the output of the serial output, output from the error log and also the output of a hardware scan from the support menu

xxyzuo, 2025-04-06:
Version 1.4.1 runs well enough to detect the temperature and gravity sensors

xxyzuo, 2025-04-06:
I think I found out why your device is not having problems, when I upgraded from 1.4.1 to 2.1.0 I could also detect the temperature and gravity sensors

mp-se, 2025-04-06:
In v2 i switched from onewire to onewireng since that lib didnt work well with esp32 boards and was not really maintained anymore. 

I can reproduce the issue since all my builds with esp8266 works fine so there must be something that is different.

What esp8266 board do you use? What tempsensor do you use? Dallas or clones?

xxyzuo, 2025-04-06:
I use Wemos D1 Mini V3 like this http://amazon.com/Internet-Development-ESP8266-Arduino-MicroPython/dp/B0CJM8GXQC
Tempsensor is DS18B20 T0-92

mp-se, 2025-04-06:
Thats the same as i have so strange its not working. Well it must be something with the onewire library. Do you have vscode and can do a build? I can give you instructions to change the libs

xxyzuo, 2025-04-07:
I'm not too good at code, so if the operation isn't too complicated, I can try to

mp-se, 2025-04-07:
No coding need, just change some dependencies and to build/flash from vscode.

1. Download and install vscode
2. Download and install Github desktop
3. Clone the gravitymon project to a local folder
4. Open vscode and open the cloned folder and let vscode install platformio and dependencies.
5. Select the target: gravity-release (for esp8266)
6. Open the file platformio.ini

7. Change the following line
``` 
https://github.com/pstolarz/OneWireNg#0.13.3
``` 
to
``` 
https://github.com/PaulStoffregen/OneWire
``` 

8. Connect your esp8266 and run the Upload command. See if that works.

mp-se, 2025-04-09:
Well then, i don't really have any ideas on what the issue could be. 

The line that you changed is the only thing that has changed compared to the 1.4.1 version that worked, so I'm not really know how this could be a software issue. It must bea hardware thing that causes this issue. I can reproduce the issue and if it would be a software issue I'm sure someone else would have reported this sooner....

My recommendation would be to get some temp sensors (and perhaps esp8266) from another source and try to ensure they are genuine and check with those.

gizmocuz, 2025-05-12:
I have exactly the same issue as above.

Compiled/Uploaded 2.2.0 and it also displays it can not find the gyro/temp sensor.

Compiling/Uploading the latest iSpindel results in a working Gyro/Temp sensor.

I also tried replacing the onewire library, but this gives the same result (no gyro/temp sensor is detected)

I have read some commits, and you are using a new type of Gyro sensor, could that have caused an issue?

Maybe iSpindel is using different settings in **void initAccel()**, or is using different order of function calls

Pins are the same (SDA/SCL D3/D4)

I tried to change a few lines, but did not succeed yet...

mp-se, 2025-05-12:
Well, the temp sensor and gyro are two different things. Gyro is using I2C for communication (or called twowire) while temp sensor is using a onewire protocol for communication. 

I'm using the CherryPhilip 4.0 PCB in my builds and I have the software working on those builds which is what an iSpindel also would use. I try to use only genuine components either Lolin D1 mini v4 and Dallas temp sensors (so that is what I use for testing). 

But in order to troubleshoot this I need some more input and there is a build in hardware test feature that you can use to see what is detected. Its under Other -> Support -> Hardware scan, run that and post the response here.

mp-se, 2025-05-12:
I just checked the ispindle code and that actually scans d1 and d6 for a tempsensor, so how is your built?

mp-se, 2025-05-13:
You can try the hardware scan using the version in the dev branch, i have added scanning of all the d1 pins for a onewire sensor to see if the sensor can be detected

gizmocuz, 2025-05-13:
Thank you for the scan function. I compiled/uploaded and monitored the dev branch, but could not see how to activate the scan.
But!! It is working now!, After importing the old iSpindel config and calibrating, all seems to be working. Only thing I had to do was to specify -10 degrees for my temperature sensor... I know it is hot here (26.9), but not 37 (celcius)
Not sure why this is

I then switched to the master branch, and this is also working!

Maybe the merge from yesterday solved something! Happy user here! Thanks for this project!!!

mp-se, 2025-05-13:
The function is found under Other -> Support -> Hardware scan, as i described above. The output is a JSON document with the scan results. 

If you have the ispindle in a tube running for a while the temperature will increase quite fast. Check the temperature outside the tube and after it has had a chance to cool off. When in gravity mode it will take 5 minutes for the sensor to cool down after the device has been opertional. A few degress in compoensation might be ok but not 10 degrees. 

I havent really done any changes to the code only some minor refactoring that should not affect the functionallity... strange that it just started working, could be some issue related to flashing the device that might have strange effects.

gizmocuz, 2025-05-16:
Thank you for your help.

I can send you the scan results if you want them? But it will take a few more days....
I noticed that the temperature is now indeed off by 6-8 degrees, the -10 was indeed not needed

mp-se, 2025-05-16:
> Thank you for your help.
> 
> I can send you the scan results if you want them? But it will take a few more days.... I noticed that the temperature is now indeed off by 6-8 degrees, the -10 was indeed not needed

please do, would be interresting to see

mp-se, 2025-05-21:
Looks like a good result, shows all the parts that is scanned for, even the pin scanning that I added works fine.

gizmocuz, 2025-05-21:
Thanks! All works perfectly now!

mp-se, 2025-05-28:
Reopening this since there are others that have similar issues which I cannot repeat on my devices. If you have issues with temperature sensor detection I have created 2 small firmwares with just temperature sensor interaction on the esp8266 using the 2 libraries that are mainly used OneWire and OneWireNG. I want to check if these are the issue or not.

1. Start the webflasher

gravitymon.com/flasher/test.html

2. Choose the debug option and try the different onewire options available. It only writes to the serial console which can be seen in the console option in the flasher. Flash the device and start the console & log function in the flasher.

3. If you don't get any serial output wait 10-15 seconds and then press reset device which should flush the serial output. you should get an output like this

```
  3574 W: Starting temperature sensor testing program
  6633 I: Found onewire sensor on pin 12.
  7764 I: Sensor 0 adr=28ac448f6783c5d, type=DS18B20
  7764 I: Found 1 temperature sensor(s).
  8259 V: Reciving temp value for DS18B20 sensor 23.50 C.
```

4. Report in this thread your findings, if one of these works then great and can adjust what library to use for better compatibility. If none then its likely a hardware or flashing issue.

gizmocuz, 2025-05-28:
I found the issue (which I had above, and still have).....

So, when the device is powered on AND connected to my Wifi (also when powered by USB), the gravity and temperature sensor is detected.

BUT... when the device is in config mode (first time, or 3 presses, each press within a second), the device cannot find the gravity or temperature sensor.

```
{
  "status": false,
  "success": true,
  "message": "",
  "i2c": [],
  "chip": {
    "family": "ESP8266"
  },
  "onewire": [
    12
  ],
  "temp_sensor": [],
  "gyro": {
    "family": ""
  }
}
```

I tested this 3 times... it does not work. When I let it power-on normally, or plug it into USB, it works:

```
{
  "status": false,
  "success": true,
  "message": "",
  "i2c": [
    {
      "adress": "0x68"
    }
  ],
  "chip": {
    "family": "ESP8266"
  },
  "onewire": [
    12
  ],
  "temp_sensor": [
    {
      "adress": "28ffa09576181e1",
      "family": "DS18B20",
      "resolution": 9
    }
  ],
  "gyro": {
    "family": ""
  }
}

```

mp-se, 2025-05-29:
The sensors are not initialized when in wifi setup mode only when in gravity or config mode. I have updated the ui for 2.2 to better reflect that part. 

Could this be what you experienced?

gizmocuz, 2025-05-30:
I triple pressed the reset button to enter config mode. The reason I did that was to be able to calibrate the device (measuring SG for the Formula)
But when in this mode, the gravity/tilt was not displayed and no sensors where detected.

When I normally start the device (turning it on), it connects to my wifi, finds the sensors, and pushes the measurements (in my case) to MQTT, then goes to sleep
Of course I can do the measurements in this mode, but it will take a long time

mp-se, 2025-05-30:
> I triple pressed the reset button to enter config mode. The reason I did that was to be able to calibrate the device (measuring SG for the Formula) But when in this mode, the gravity/tilt was not [...]

Pressing the reset button will not get you into configuration mode, only wifi setup mode. Place the device flat on a floor or charge it, remove the battery or connect the tx/rx pins during startup will get you into config mode.

gizmocuz, 2025-05-30:
I don't want to go to config mode (probably when you choose to reset the device to factory in setup you get this too)

I want to calibrate my define in a bucket of water with sugar to generate the formula.

Triple clicking lets me enter the web interface, where (i hope) should see the tilt value, which I can enter in the forumula.

But the gravity/temp sensor is not detected

mp-se, 2025-05-30:
> I don't want to go to config mode (probably when you choose to reset the device to factory in setup you get this too)
> I want to calibrate my define in a bucket of water with sugar to generate the [...]

Well, using configuration mode to read the sensor and do calibration is the way its designed to be used. Wifi mode is only there to set the wifi settings. When you tripple click you get into wifi mode and the sensors are not enabled then, this is by design.

Place the device on a flat surface (gyro > 90 degrees), do a reset, open the web interface and toggle "force config mode" (on older versions it will state "force gravity mode" which was a bug. This will stop the device from entering sleep mode. 

When that is turned on you can start the calibration sequence and the device will never go to sleep. You will get an average angle that you can reset if you want to get a better reading. You can also switch to the calibration page and enter the values directly and let the device propose/generate matching formulas for your data

gizmocuz, 2025-06-04:
Enabling 'Force config mode' does not enable the sensors, and pressing reboot will just connect to the wifi, and when the device lays still for a few seconds, it turns off (and waits 10 minutes before transmitting data)

gizmocuz, 2025-06-04:
No this is way more complex. And... as mentioned above, after a while it goes to sleep.

I need to walk back and forward to a bucket with water/sugar, measure SG, then check in the web interface the tilt. and put that in the calculation sheet. And I need to do this multiple times.
"Then" I calculate the formula.

What is the reason the sensors are 'disabled' when you triple press reset button? It also does not match the online video's and description.

Why is there a 'config' mode, and a 'wifi/setup' mode? why not 1 config mode and 1 running mode?

mp-se, 2025-06-04:
Well the online video is for the ispindel software not for gravitymon so the video would not apply to how you use gravitymon.

gizmocuz, 2025-06-04:
True, but you understand it is difficult for a user to get into config-mode (what is this?) and very easy in wifi-mode (ain't this just normal setup mode?).
Why are the sensors disabled in wifi-setup-mode? I can setup way more then wifi in wifi-setup mode, so is this not normal setup-mode? Is there anything different in config-mode then setup-mode? (except the sensors are working in config-mode) ?

mp-se, 2025-06-04:
If there is an issue with the hardware or sensors that causes a crash/startup issue the only way to get into the device is through wifi setup mode and update settings or do a firmware update. Since this is also used as failsafe mode its not desired to enable the sensors then.

gizmocuz, 2025-06-04:
Okey understood. Thanks for you patience!
I have now calibrated my spindle and can't wait to test it next week!