mp-se/gravitymon #226: GravityMon freezes in gravity mode after wakeup (FLOATY HW)

sigma415, 2025-05-26:
**Describe the bug**

GravityMon freezes in gravity mode after first wakeup from sleep mode.
Last output in serial monitor is "283 I: GYRO: Restored MPU6050/MPU6000".
After this no more reaction, only reset of ESP32 revives the device.

I have built up the Floaty pill hardware from DIYhomebrewers.com which runs on the ESP32 Lolin lite with MPU6050 gyro supplied by GPIO pins 33 and 25.
For testing I did not desolder red LED on gyro board so I can see if it's powered or not.

**To Reproduce**
1. Start up in 90 degree (calibrated) position with "Force config mode" = Enabled. Anything runs fine and also gravity and temperature data is transmitted via BLE
2. Tilt the board into gravity mode position and disable "Force config mode" (or in opposite order, doesn't matter)
3. device goes to sleep: red gyro LED switches off and also blue blinking Lolin LED switches off
4. after sleep time (in my case only 10 seconds for testing) gyro LED switches ON again, blue Lolin LED stays off
5. that's it ! No more reaction ....

**Expected behavior**
GracvityMon device wakes up after 10 sec sleep time, transmits SG data via BLE and goes to sleep again for the next period and so on.

**Screenshots**
I will add the serial monitor output file (Putty) and also the GravityMon_HW_scan.json

**Software:**
 - Platform: ESP32lite Lolin32 (Floaty HW)
 - Version: GravityMon v2.2.0-beta3, also tried v2.1.0 Github version with Floaty fix

**Additional context**
I'm not sure, if this a SW bug or maybe still a problem with my HW, but I'm an electronics engineer and this is not my first ESP32 project. I have also already tried several HW measures like VCC decoupling capacitor w/o success.

@mp-se: Thanks a lot for your support  :)

[putty_20250526_131624_COM8_0.log](https://github.com/user-attachments/files/20446526/putty_20250526_131624_COM8_0.log)

[GravityMon_HW_scan.json](https://github.com/user-attachments/files/20446527/GravityMon_HW_scan.json)

sigma415, 2025-05-27:
Bingo !
With 2.2.0 beta1 it works like it should :)

I attach the (working) log anyway.
Please let me know, if I can help doing some debugging or testing with my Floaty HW.

[putty_20250527_125212_COM8_0_2.2.0_beta1_working.log](https://github.com/user-attachments/files/20455521/putty_20250527_125212_COM8_0_2.2.0_beta1_working.log)

mp-se, 2025-05-27:
Ok, it could be the RTM memory part that is now working well on the ESP32 lite, I will try to remove that part and see if that works. Thanks that helps with troubleshooting.

sigma415, 2025-05-30:
Thank you. 
Only to double check: You mean "Commit 81fa9d1", right ?
https://github.com/mp-se/gravitymon/commit/81fa9d1a6ea960af40c9c8a1768e94d288e29fce

I will test.

mp-se, 2025-05-31:
Just grab the latest version in the dev branch and you will be fine.

sigma415, 2025-06-01:
BTW: CoPilot AI gave me this answer for my question: "Does ESP21lite has other RTC memory ?"

-------------------------------

Copilot said: 

It seems like you are asking whether the ESP32 Lolin Lite (ESP32lite) board has a different kind of RTC (Real-Time Clock) memory compared to other ESP32 variants, possibly in the context of the GravityMon project.

ESP32 RTC Memory Overview:

    All ESP32 chips (including ESP32 Lolin Lite) have two types of RTC memory:
        RTC Fast Memory (8 KB): Retained during deep sleep if enabled, used for code and variables that must survive deep sleep.
        RTC Slow Memory (8 KB): Also retained during deep sleep, typically used for storing data accessible by the ULP (Ultra Low Power co-processor) and for variables that must persist across deep sleep cycles.

RTC Memory in ESP32lite (Lolin32 Lite):

    The ESP32 Lolin Lite uses the standard ESP32-WROOM-32 module or similar, so it has the same RTC memory features as other ESP32 boards.
    There is no "other" or special RTC memory compared to other ESP32 boards. However, how RTC memory is used or mapped can depend on board design, firmware, or power management features.

Important Notes:

    If you are experiencing issues with wake-from-sleep and memory retention (as in your GravityMon issue), the problem is likely in firmware (how variables are placed in RTC memory, power domain control, or how the chip is re-initialized) or hardware power supply stability, not in a difference in the RTC memory itself.
    Always ensure you use RTC_DATA_ATTR or RTC_NOINIT_ATTR attributes in your code to place variables in RTC memory.

Summary Table:
Board	RTC Fast Memory	RTC Slow Memory	Retained after Deep Sleep
ESP32 Lolin Lite	8 KB	8 KB	Yes (if used properly)
ESP32 (generic)	8 KB	8 KB	Yes (if used properly)

References:

    [ESP-IDF docs: RTC Memory](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/deep-sleep-stub.html#using-rtc-memory)
    [Arduino-ESP32 docs: Deep Sleep](https://docs.espressif.com/projects/arduino-esp32/en/latest/deep-sleep.html#rtc-memory)

Let me know if you want help with code examples or debugging RTC memory usage!

mp-se, 2025-06-05:
Well as you state the device is quite old at this stage so if the version works i don't really want to put more effort into it. This version might be the last version for the floaty design since i dont have the hardware to test with. 

There are many newer ispindle based designs that works very well

sigma415, 2025-06-05:
> Well as you state the device is quite old at this stage so if the version works i don't really want to put more effort into it. This version might be the last version for the floaty design since i dont have the hardware to test with.

Does this mean you will release a new (and last for Floaty) 2.2 version with the "state keeping" changes (gpio_deep_sleep_hold_en();) for the original Floaty design ?

> There are many newer ispindle based designs that works very well

Could you recommend a good iSpindel HW design with ESP32-S3 and wireless charging ?

mp-se, 2025-06-05:
I will release the version i have now as 2.2 as soon as i have had the chance to test it some more.

I will probably leave the floaty code in there and treat it as a custom board for now.

I dont really have a recommendation for a hardware build, but i have started to collaborate with one user to see if he can adapt his design for the esp32 which include wireless charging. 

You can see his project here, https://github.com/hobipivo/iSpindel

This is another project that is working on an adaption https://github.com/dtashmore547 which there is a thread in the discussions

sigma415, 2025-06-05:
Thank you.
I have found your interesting discussion with @hobipivo here: https://github.com/hobipivo/iSpindel/issues/2#issue-3086073911

Up to now there's nothing to see about wireless charging but I understood in this discussion that @hobipivo wants to go this direction.
Also I could not find any KiCad files of his PCB's in his repository, so his project seems not to be open source.
Maybe I should also to start my own circuit and PCB design for a solution with RAPTpill case, ESP32, BLE, GravityMon and wireless charging....  ;)

closed 2025-06-08 by mp-se