mp-se/gravitymon #230: Add support for ESP32 zero boards

mp-se, 2025-06-08:
Look into support for ESP zero / super mini boards that much smaller form factors. Example of boards to consider adding: 

The challenge is that many super mini boards exist and not all have the same pin configuration. In order officially support these boards I need to ensure the correct pinout which means that I will focus on known brands.

# Waveshare ESP32 C3 zero

Pins: 5V, GND, 3.3V, 0, 1, 2, 3, 4, 5, 6 | 27, 8, 9, 10, 11, 12, 13, RX, TX

[image]

# Waveshare ESP32 C6 zero

Pins: 5V, GND, 3.3V, 0, 1, 2, 3, 4, 5 | 22, 21, 20, 19, 18, 15, 14, RX, TX

[image]

# TENSTAR ESP32 C3 zero 

Pins: 5V, GND, 3.3V, 0, 1, 2, 3, 4, 5, 6 | 27, 8, 9, 10, 11, 12, 13, RX, TX

[image]

hobipivo, 2025-07-12:
Are modules with dual-core processors required for the iSpindle?

When do you need to switch to Adriano-3 (8MB modules)?

The ESP32 Super Mini modules should be easy to find and cheap.
I would prefer an ESP32 C3 Super Mini (Zero) 16-pin module with layout: **5, 6, 7, 8, 9, 10, 20,21 | 0, 1, 2, 3, 4, 3V, GND, 5V** for the first step.
https://dl.artronshop.co.th/ESP32-C3%20SuperMini%20datasheet.pdf

Maybe there will be more ESP32 Super Mini Modules with 8MB Flash in the future.

mp-se, 2025-07-12:
Dual core is not essential, i'm only using one core currently. But two cores could speed up the sensor readings if more could be done in parallel. One thread for connecting to wifi and the other for reading sensor values. But since I need to support 8266 its easier to do everything in one thread. 

I'm currently on 1.8 Mb with the current features and when i have tested with Arduino 3 it added 300k on top of that which means I can do flashing via web or OTA. So to move to Arduino 3 i would need 8Mb RAM.

I ordered a few boards from aliexpress and I focused on getting some brand ones since I want to make it easy for users to find the correct board. Some of the super mini does not have the same pinout which means the software would not work. 

I ordered a the TENSTAR and WAVESHARE board but i did not receive the 16pin TENSTAR as in the picture above, instead i got a 18pin board that was identical to the waveshare except for the print on the back. 

Currently I'm testing those boards to see if they can be used even though these are 18-pins. 

If you have some brand 16 pin boards that could be recommended i would order the try those as well. The ones i have look at had different pinouts so it would be hard to find the correct ones.

mp-se, 2025-07-12:
This is the current schematic im testing with the Tenstar/Waveshare Zero board. I've just done some initial hardware test and next step is to make a prototype to do some more testing with. 

I have also added a storage pin feature that will place the board in deep sleep when more than 2.5V is applied to the pin. The intention is to use that when a wireless charger is connected so the board will go to sleep while charging. Once power is removed the device will wakeup. No real need for a power on/off switch. 

[image]

mp-se, 2025-07-13:
Support for the ESP32 c3 Zero from Waveshare / Tenstar (18-pin board) is now available in the dev branch. More testing is needed but the basic tests has passed.

mp-se, 2025-07-13:
> Are ESP32 modules with dual-core processors required for the iSpindle?
> 
> When do you need to switch to Adriano-3 (8MB modules)?
> 
> The ESP32 Super Mini modules should be easy to find and cheap. I would prefer an ESP32 C3 **Super Mini** (Zero) 16-pin 4MB module with layout: **5, 6, 7, 8, 9, 10, 20, 21 | 0, 1, 2, 3, 4, 3V, GND, 5V** for the first step. https://dl.artronshop.co.th/ESP32-C3%20SuperMini%20datasheet.pdf
> 
> Maybe there will be more ESP32 Super Mini Modules with 8MB Flash in the future.

I have placed a new order for an ESP32 super mini so lets see if I can get the correct one this time. I have added support for the waveshare board so that seams to work fine.

hobipivo, 2025-07-14:
> Support for the ESP32 c3 Zero from Waveshare / Tenstar (18-pin board) is now available in the dev branch. More testing is needed but the basic tests has passed.

I have ordered the TENSTAR ESP32 C3 Zero (18-pin board).
The “Storage” input for charging with Qi Charger should be renamed to “Charging” as it will also be active when charging with USB-C.
What does the solution for “force config” look like?

mp-se, 2025-07-19:
> I have ordered the TENSTAR ESP32 C3 Zero (18-pin board). The “Storage” input for charging with Qi Charger should be renamed to “Charging” as it will also be active when charging with USB-C. What [...]

Thats a good idea, I will change the naming to Charging PIN

Force config can be set either via the UI or by connecting the two pins CFG1/CGF2 which is checked at startup and sets the option on the device. As long as that is set it will not go into sleep mode. This is useful when doing calibration and getting gravity/angle values for the calibration formula.

mp-se, 2025-07-25:
I can see that the boards wants to wake up (connects to serial but does not complete the boot sequence) my guess its must be some of the strapping pins that causes this. I dont have a soldering iron where I'm at the moment but my best guess is the GPIO8 / GPIO9 that should be high during boot but on my board they are low.. So there might not be any pullup resistors on the gyro board. 

Probably its best to move the i2c to some other pins to avoid this or add some pullup resistors to the SCL / SDA lines. 

I will do some more testing when I can resolder my test board.

mp-se, 2025-07-29:
So, here is an update schema of the ESP32c3 Zero where i have changed some pins to avoid the strapping pins. This version is in the dev branch.

[image]

hobipivo, 2025-07-30:
After installing the new firmware version under branch/dev, the following message appears in the browser for both module variants C3 Zero and C3 Super Mini:
Failed to load status from device, please try to reload page!

mp-se, 2025-08-02:
The build in the dev branch should be good now.

mp-se, 2025-08-05:
> : Setting up hardware.
> 394 I: GYRO: Detected MPU6050/MPU6500 68.
> 1283 I: Main: Battery 3.89 V, Gyro=88.49, Run-mode=1.
> 1326 I: WIFI: Reducing wifi power for c3 chip.
> 1327 I: WIFI: Connecting to wifi using stored settings Zeluce33.
> 7335 I: WIFI: Found the following networks ssid: Zeluce33 bssid: 74:42:7F:3F:21:16, rssi: -81 channel: 6.
> 7336 I: WIFI: Found the following networks ssid: Zeluce33 bssid: A0:04:60:37:70:83, rssi: -86 channel: 1.
> 7337 I: WIFI: Using ssid: Zeluce33 bssid: 74:42:7F:3F:21:16, rssi: -81 channel: 6.
>   ..

> ```
> : Setting up hardware.
> 394 I: GYRO: Detected MPU6050/MPU6500 68.
> 1282 I: Main: Battery 3.85 V, Gyro=89.11, Run-mode=1.
> 1324 I: WIFI: Reducing wifi power for c3 chip.
> 1325 I: WIFI: Connecting to wifi using stored settings Zeluce33.
> 7333 I: WIFI: Found the following networks ssid: Zeluce33 bssid: 74:42:7F:3F:21:16, rssi: -85 channel: 6.
> 7334 I: WIFI: Found the following networks ssid: Zeluce33 bssid: A0:04:60:37:70:83, rssi: -89 channel: 1.
> 7335 I: WIFI: Using ssid: Zeluce33 bssid: 74:42:7F:3F:21:16, rssi: -85 channel: 6.
>                                          

From what i can see the error comes from the arduino framework and my guess is that in the first attempt the signal is weak (-81) and in the second -85 which is weaker and barley connectable which is proably why it fails the second time. The c3 boards have had quite bad Wifi reception where i was forced to lower the transmission power for them to work. It might be that the newer boards have fixed that and I can remove that limitation.

mp-se, 2025-08-24:
I have not done some initial testing with the zero and super mini boards and the basics seams to work. I need to make a full prototype to test all the features. The dev branch is up to data with the latest changes.

hobipivo, 2025-08-26:
With “firmware32c3zero.bin **beta-2**,” I get the error “**battery level not valid**” on both Zero boards.
The voltage at pin 4 is 1.75V and works with the previous version “firmware32c3zero.bin beta.”

mp-se, 2025-08-26:
ok, i will have a look, i just tested the pin configuration for the other features, i have not tried the battery not the sleep/charging features in a complete build. 

I'm still waiting for some new components so I can build a few new prototypes and test everything properly

mp-se, 2025-08-26:
I tested the supermini and the voltage reading looks good. What voltage factor are you using. I calibrated mine to 1.78 (the standard for the c3 mini was 2.25) which is to high.

mp-se, 2025-08-26:
You where correct there was a bug affecting the zero board. It should be fixed now (build will be ready in 15 min). I have also added the default voltage factors that I calculated for those boards.