mp-se/gravitymon #231: Wifi Setup mode instead of Storage sleep when <5 degree

sigma415, 2025-06-14:
**Describe the bug**
Instead of going to a 2000 minutes sleep device goes to Wifi Setup mode (LED always on).
Only way to get out: Connect to AP -> select Wifi setting > Restart

**To Reproduce**
Steps to reproduce the behavior:
1. Enable "Storage Sleep"
2. Enable "Force Config Mode"
3. Place device on its cap (Angle shows 3.23 degree)
4. Disable "Force Config Mode"  (step 3 & 4 can also be reversed)
5. After some seconds blue LED turns always on, device has entered Wifi Setup mode
6. Only (elegant) way to get out: Connect to AP -> select Wifi setting > Restart

**Expected behavior**
Devive shoukd go to sleep for 2000 minutes or so ...

**Software:**
 - HW: ESP32Lite (Floaty), but with gyro module directly supplied by +3V3 and GND (not via GPIO pins. See closed issue with RTC memory)
 - Version: 2.2.0 (rc-1) UI: 2.2.0 (..533e61)   (but same behaviour with beta3)

**Additional context**
I know you want to stop ESP32Lite support, but maybe other platforms are also affected.

mp-se, 2025-06-14:
Strange, it should not go into wifi setup mode unless the double reset detector is triggered. I will add a call to reset that counter before it goes to deep sleep. I would highly recommend to add a reed switch to the reset pin if you are using deep sleep feature so you can wake it up when you need to. 

I have just pushed the fix to dev and it will probably be ready in 10 minutes after the automated build.

sigma415, 2025-06-14:
I have had a reed contact on EN pin to GND, but it was a bad idea on the ESP32Lite.
The reed contact welded because there is a 1uF capacitor directly on EN of ESP32Lite which results in a high discharge current spike. After renewing the reed and added a 33R in series I got multiple resets which also ended in wireless setup mode ...

sigma415, 2025-06-15:
Magnus, thank you. (couldn't test yesterday)
I can confirm, that `storage sleep` works with your dev version from 2025-06-14 [firmware32lite.bin](https://github.com/mp-se/gravitymon/blob/dev/bin/firmware32lite.bin) 
And I experienced the nice behaviour, that tilting back to 90deg during `storage sleep` goes immediately (after one or two seconds) back to `configuration mode`. 
I miss this behaviour in normal `Gravity mode`. There one must wait until the `Sleep Intervall` defined in `Push - Settings `times out.
I would have expected, that tilting back to 90deg in `Gravity mode` also immediately goes to `Configuration mode` without waiting.
Are there different "sleeps" with and w/o gyro monitoring ?

Another (minor) issue:  Connecting the TX/RX pins (for Floaty design GPIO13 and 15) also will **not** force the device into `configuration mode`. Nothing happens when connected (up to now in all GM versions I have flashed)

Thank you for your support and patience.

Greetings from Germany
Alex

closed 2025-06-17 by mp-se

mp-se, 2025-06-17:
Closing this since the fix adressed the issue