mp-se/gravitymon #235: Unnable Acess to web ui in config mode

GabrielPeral, 2025-07-01:
Hello,
First want to say thanks a lot for all your effort, this tool looks amazing, second sorry if this is not the right place to write this.

I have installed for the first time your newest version 2.2.0 in a LOLIN32 CH340 board, I already have installed the gyro (GY-521 MPU-6050) and the resistor for the battery level.

However, seems like once I flashed the firmware using this URL (https://web.brewflasher.com/) and then setting the Wi-Fi (otherwise it gives me error on scan in the UI (but console finds the Wi-Fi)) here https://www.gravitymon.com/flasher/index.html I can then access the portal, configure name, MQTT, and other stuff (also got an IP from my network)

However, the moment I reset the device, I am no longer able to access the UI, It gives me this error:

[image]

And here are the logs of the console:
 ```
 412692 I: WEB : webServer callback for /api/config/format(read).
[412696][E][vfs_api.cpp:105] open(): /littlefs/http-1.tpl does not exist, no permits for creation
[13:08:44][412711][E][vfs_api.cpp:105] open(): /littlefs/http-2.tpl does not exist, no permits for creation
[13:08:44][412723][E][vfs_api.cpp:105] open(): /littlefs/http-3.tpl does not exist, no permits for creation
[13:08:44][412736][E][vfs_api.cpp:105] open(): /littlefs/influxdb.tpl does not exist, no permits for creation
[13:08:44]Guru Meditation Error: Core  1 panic'ed (Double exception). 
[13:08:44]
[13:08:44]Core  1 register dump:
[13:08:44]PC      : 0x4009766a  PS      : 0x00040636  A0      : 0x80083308  A1      : 0x3ffdc260  
[13:08:44]A2      : 0x00000000  A3      : 0x00000001  A4      : 0x3ffbdf28  A5      : 0x00000000  
[13:08:44]A6      : 0x3ffdc560  A7      : 0x00000004  A8      : 0x40080080  A9      : 0x3ffdc370  
[13:08:44]A10     : 0x00060836  A11     : 0x00040026  A12     : 0x00000000  A13     : 0x3ffbf808  
[13:08:44]A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000017  EXCCAUSE: 0x00000002  
[13:08:44]EXCVADDR: 0xffffffe0  LBEG    : 0x40092efc  LEND    : 0x40092f07  LCOUNT  : 0x00000000  
[13:08:44]
[13:08:44]
[13:08:44]Backtrace: 0x40097667:0x3ffdc260 0x40083305:0x3ffdc280 0x40097667:0x3ffdc2a0 0x40097667:0x3ffdc2c0 0x40097667:0x3ffdc2e0 0x40097667:0x3ffdc300 0x40097667:0x3ffdc340 0x40097667:0x3ffdc370 0x4008007d:0x3ffdc260 0x400841d1:0x3ffdc280 0x40097667:0x3ffdc2a0 0x40097667:0x3ffdc2c0 0x40097667:0x3ffdc2e0 0x40097667:0x3ffdc300 0x40097667:0x3ffdc340 0x40097667:0x3ffdc370 0x4008007d:0x3ffdc260 0x400841d1:0x3ffdc280 0x40097667:0x3ffdc2a0 0x40097667:0x3ffdc2c0 0x40097667:0x3ffdc2e0 0x40097667:0x3ffdc300 0x40097667:0x3ffdc340 0x40097667:0x3ffdc370 0x4008007d:0x3ffdc260 0x400841d1:0x3ffdc280 0x40097667:0x3ffdc2a0 0x40097667:0x3ffdc2c0 0x40097667:0x3ffdc2e0 0x40097667:0x3ffdc300 0x40097667:0x3ffdc340 0x40097667:0x3ffdc370 0x4008007d:0x3ffdc260 0x400841d1:0x3ffdc280 0x40097667:0x3ffdc2a0 0x40097667:0x3ffdc2c0 0x40097667:0x3ffdc2e0 0x40097667:0x3ffdc300 0x40097667:0x3ffdc340 0x40097667:0x3ffdc370 0x4008007d:0x3ffdc260 0x400841d1:0x3ffdc280 0x40097667:0x3ffdc2a0 0x40097667:0x3ffdc2c0 0x40097667:0x3ffdc2e0 0x40097667:0x3ffdc300 0x40097667:0x3ffdc340 0x40097667:0x3ffdc370 0x4008007d:0x3ffdc260 0x400841d1:0x3ffdc280 0x40097667:0x3ffdc2a0 0x40097667:0x3ffdc2c0 0x40097667:0x3ffdc2e0 0x40097667:0x3ffdc300 0x40097667:0x3ffdc340 0x40097667:0x3ffdc370 0x4008007d:0x3ffdc260 0x400841d1:0x3ffdc280 0x40097667:0x3ffdc2a0 0x40097667:0x3ffdc2c0 0x40097667:0x3ffdc2e0 0x40097667:0x3ffdc300 0x40097667:0x3ffdc340 0x40097667:0x3ffdc370 0x4008007d:0x3ffdc260 0x400841d1:0x3ffdc280 0x40097667:0x3ffdc2a0 0x40097667:0x3ffdc2c0 0x40097667:0x3ffdc2e0 0x40097667:0x3ffdc300 0x40097667:0x3ffdc340 0x40097667:0x3ffdc370 0x4008007d:0x3ffdc260 0x400841d1:0x3ffdc280 0x40097667:0x3ffdc2a0 0x40097667:0x3ffdc2c0 0x40097667:0x3ffdc2e0 0x40097667:0x3ffdc300 0x40097667:0x3ffdc340 0x40097667:0x3ffdc370 0x4008007d:0x3ffdc260 0x400841d1:0x3ffdc280 0x40097667:0x3ffdc2a0 0x40097667:0x3ffdc2c0 0x40097667:0x3ffdc2e0 0x40097667:0x3ffdc300 0x40097667:0x3ffdc340 0x40097667:0x3ffdc370 0x4008007d:0x3ffdc260 0x400841d1:0x3ffdc280 0x40097667:0x3ffdc2a0 0x40097667:0x3ffdc2c0 0x40097667:0x3ffdc2e0 0x40097667:0x3ffdc300 0x40097667:0x3ffdc340 0x40097667:0x3ffdc370 0x4008007d:0x3ffdc260 0x400841d1:0x3ffdc280 0x40097667:0x3ffdc2a0 0x40097667:0x3ffdc2c0 0x40097667:0x3ffdc2e0 |<-CONTINUES
[13:08:44]
[13:08:44]
[13:08:44]
[13:08:44]
[13:08:44]ELF file SHA256: e998a5ae94dc95f1
[13:08:44]
[13:08:44]E (1628) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
[13:08:44]Rebooting...
[13:08:44]ets Jun  8 2016 00:22:57
[13:08:44]
[13:08:44]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[13:08:44]configsip: 0, SPIWP:0xee
[13:08:44]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:08:44]mode:DIO, clock div:2
[13:08:44]load:0x3fff0018,len:4
[13:08:44]load:0x3fff001c,len:1044
[13:08:44]load:0x40078000,len:10044
[13:08:44]load:0x40080400,len:5872
[13:08:44]entry 0x400806ac
[13:08:45]Serial console activated.
[13:08:45]         4 I: SDBG: Serial logging started at 115200.
E (699) esp_core_dump_flash: ���ɕ�dump partition found!
[13:08:45]E (699) esp_core_dump_flash: No core dump partition found!
[13:08:45]        41 I: Main: Started setup for 0f4c70.
        42 I: Build options: 2.2.0 (..2afa94) LOGLEVEL 5 GRAVITYMON ESP32 FLOATY 
        43 I: UTIL: Chip=ESP32, Rev=1, Feat=0x0032
        75 I: WIFI: Current reset counter 0.
        98 I: UTIL: Last reset cause 'software reset CPU' (12)
[   102][E][vfs_api.cpp:105] open(): /littlefs/gravitymon.json does not exist, no permits for creation
[13:08:45][   116][E][vfs_api.cpp:105] open(): /littlefs/hardware.json does not exist, no permits for creation
[13:08:45]       215 I: CFG : Configuration file loaded.
       218 I: CFG : Using wifi settings from EEPROM.
[   234][E][vfs_api.cpp:105] open(): /littlefs/drd.dat does not exist, no permits for creation
```
Is this normal?? I see the LED blinking blue so it is on config mode, I've flashed the firmware several times and is always the same behavior

Thanks again for all your job

Regards,

Gabriel

GabrielPeral, 2025-07-01:
it shows this:

[image]

and the output of the terminal:
```
[18:37:21]     80816 W: Main: Angle: 79.79 (79.79), Velocity: 0.00
     81717 W: Main: Angle: 79.79 (79.79), Velocity: 0.00
     82618 W: Main: Angle: 79.78 (79.78), Velocity: 0.00
     83432 I: WEB : webServer callback for /api/ping.
     83521 W: Main: Angle: 79.79 (79.79), Velocity: 0.00
     84421 W: Main: Angle: 79.80 (79.80), Velocity: 0.00
     85322 W: Main: Angle: 79.79 (79.79), Velocity: 0.00
     86223 W: Main: Angle: 79.78 (79.78), Velocity: 0.00
     87124 W: Main: Angle: 79.77 (79.77), Velocity: 0.00
     88025 W: Main: Angle: 79.78 (79.78), Velocity: 0.00
     88926 W: Main: Angle: 79.78 (79.78), Velocity: 0.00
     89827 W: Main: Angle: 79.84 (79.84), Velocity: 0.00
     90088 I: WEB : webServer callback for /api/ping.
     90729 W: Main: Angle: 79.84 (79.84), Velocity: 0.00
     91629 W: Main: Angle: 79.83 (79.83), Velocity: 0.00
     92531 W: Main: Angle: 79.84 (79.84), Velocity: 0.00
     93431 W: Main: Angle: 79.84 (79.84), Velocity: 0.00
     94333 W: Main: Angle: 79.84 (79.84), Velocity: 0.00
     95208 I: WEB : webServer callback for /index.html (Memory).
     95234 W: Main: Angle: 79.82 (79.82), Velocity: 0.00
[ 95508][E][vfs_api.cpp:105] open(): /littlefs/app.js.gz does not exist, no permits for creation
[18:37:22]     95517 I: WEB : webServer callback for /app.js (Memory).
     95529 I: WEB : webServer callback for /app.css (Memory).
     95723 W: WEB : GET on /css/bootstrap.css.map not recognized.
     95760 I: WEB : webServer callback for /index.html (Memory).
     96135 W: Main: Angle: 79.84 (79.84), Velocity: 0.00
     96230 I: WEB : webServer callback for /api/auth.
     96238 I: WEB : webServer callback for /favicon.ico (Memory).
     96434 I: WEB : webServer callback for /api/status(get).
[ 96442][E][vfs_api.cpp:105] open(): /littlefs/gravitymon.json does not exist, no permits for creation
[18:37:22]     96520 I: WEB : webServer callback for /favicon.ico (Memory).
     96531 I: WEB : webServer callback for /api/config(read).
     96939 I: WEB : webServer callback for /api/config/format(read).
[ 96943][E][vfs_api.cpp:105] open(): /littlefs/http-1.tpl does not exist, no permits for creation
[18:37:22][ 96958][E][vfs_api.cpp:105] open(): /littlefs/http-2.tpl does not exist, no permits for creation
[18:37:22][ 96971][E][vfs_api.cpp:105] open(): /littlefs/http-3.tpl does not exist, no permits for creation
[18:37:22][ 96983][E][vfs_api.cpp:105] open(): /littlefs/influxdb.tpl does not exist, no permits for creation
[18:37:23]Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
[18:37:23]Debug exception reason: Stack canary watchpoint triggered (async_tcp) 
[18:37:23]Core  1 register dump:
[18:37:23]PC      : 0x4009a817  PS      : 0x00060436  A0      : 0x8009a2ac  A1      : 0x3ffdd740  
[18:37:23]A2      : 0x3ffbf808  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060423  
[18:37:23]A6      : 0x00060420  A7      : 0x0000abab  A8      : 0xb33fffff  A9      : 0xffffffff  
[18:37:23]A10     : 0x00000001  A11     : 0x3ffb6ce4  A12     : 0x3ffb6c98  A13     : 0x00000000  
[18:37:23]A14     : 0x007bf808  A15     : 0x003fffff  SAR     : 0x00000018  EXCCAUSE: 0x00000001  
[18:37:23]EXCVADDR: 0x00000000  LBEG    : 0x40092efc  LEND    : 0x40092f07  LCOUNT  : 0x00000000  
[18:37:23]
[18:37:23]
[18:37:23]Backtrace: 0x4009a814:0x3ffdd740 0x4009a2a9:0x3ffdd780 0x400832ec:0x3ffdd7a0 0x40083179:0x3ffdd7d0 0x40083183:0x3ffdd7f0 0x40091c0d:0x3ffdd810 0x40082d12:0x3ffdd830 0x4011827b:0x3ffdd870 0x4014744e:0x3ffdd8a0 0x40147721:0x3ffdd8d0 0x4014887a:0x3ffdd910 0x40148f4a:0x3ffdd9b0 0x4014b0d9:0x3ffdda10 0x401468b2:0x3ffdda60 0x4011e7a9:0x3ffddb90 0x40092bed:0x3ffddbb0 0x40092c59:0x3ffddc10 0x4009366e:0x3ffddc40 0x401c3d3e:0x3ffddc60 0x401c3dd1:0x3ffddca0 0x401b8e37:0x3ffddcc0 0x400e744f:0x3ffddce0 0x400df3d7:0x3ffddd00 0x400df86d:0x3ffde280 0x401d845d:0x3ffde330 0x400ed4d2:0x3ffde350 0x401d8973:0x3ffde390 0x400e8e65:0x3ffde3b0 0x400e8b8a:0x3ffde3d0 0x400e8cfa:0x3ffde3f0 0x400e8eb6:0x3ffde480 0x400e8b8a:0x3ffde4c0 0x400e8cfa:0x3ffde4e0 0x400e9516:0x3ffde570 0x400ea446:0x3ffde5b0 0x400eb105:0x3ffde5f0 0x400eb301:0x3ffde620 0x401b7c8d:0x3ffde640 0x401b7d05:0x3ffde670 0x401b87d2:0x3ffde690
[18:37:23]
[18:37:23]
[18:37:23]
[18:37:23]
[18:37:23]ELF file SHA256: e998a5ae94dc95f1
[18:37:23]
[18:37:23]E (7881) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
[18:37:23]Rebooting...
[18:37:23]ets Jun  8 2016 00:22:57
[18:37:23]
[18:37:23]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[18:37:23]configsip: 0, SPIWP:0xee
[18:37:23]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[18:37:23]mode:DIO, clock div:2
[18:37:23]load:0x3fff0018,len:4
[18:37:23]load:0x3fff001c,len:1044
[18:37:23]load:0x40078000,len:10044
[18:37:23]load:0x40080400,len:5872
[18:37:23]entry 0x400806ac
[18:37:23]Serial console activated.
[18:37:23]         4 I: SDBG: Serial logging started at 115200.
E (698) esp_core_dump_flash: ~��ɕ�dump partition found!
[18:37:23]E (699) esp_core_dump_flash: No core dump partition found!
[18:37:23]        41 I: Main: Started setup for 0f4c70.
        42 I: Build options: 2.2.0 (..2afa94) LOGLEVEL 5 GRAVITYMON ESP32 FLOATY 
        43 I: UTIL: Chip=ESP32, Rev=1, Feat=0x0032
        72 I: WIFI: Current reset counter 0.
        92 I: UTIL: Last reset cause 'software reset CPU' (12)
[    95][E][vfs_api.cpp:105] open(): /littlefs/gravitymon.json does not exist, no permits for creation
[18:37:23][   109][E][vfs_api.cpp:105] open(): /littlefs/hardware.json does not exist, no permits for creation
[18:37:24]       196 I: CFG : Configuration file loaded.
       199 I: CFG : Using wifi settings from EEPROM.
[   214][E][vfs_api.cpp:105] open(): /littlefs/drd.dat does not exist, no permits for creation
```

mp-se, 2025-07-02:
Looks like it crashes when trying to read from the flash disk so the UI dont get all the data that is needed, thats why you get the error. Could be a hardware error or defective board if you have tried to do a full erase and reflash, do you have another board to try with ?

mp-se, 2025-07-02:
Does any of the older versions work? If not then is probably bad hardware

mp-se, 2025-07-02:
Well I will drop support for that board since that is really old. I would recommend to use one of the other esp32 mini boards since they are newer. I dont have a working copy if that board any more so I cant test and validate. 

I have had users that have used the latest version so it should work even though I cant test it myself anymore.

GabrielPeral, 2025-07-14:
Got this at boot

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10044
load:0x40080400,len:5872
entry 0x400806ac
Serial console activated.
         4 I: SDBG: Serial logging started at 115200.
E (698) esp_core_dump_flash: ~�.�ɕ�dump partition found!
E (699) esp_core_dump_flash: No core dump partition found!
        39 I: Main: Started setup for 0f4d50.
        40 I: Build options: 2.2.1 (..19953f) LOGLEVEL 5 GRAVITYMON ESP32 FLOATY 
        42 I: UTIL: Chip=ESP32, Rev=1, Feat=0x0032
        83 I: WIFI: Current reset counter 0.
       300 I: UTIL: Last reset cause 'software reset CPU' (12)
[   302][E][vfs_api.cpp:105] open(): /littlefs/gravitymon.json does not exist, no permits for creation
[   316][E][vfs_api.cpp:105] open(): /littlefs/hardware.json does not exist, no permits for creation
       367 I: CFG : Configuration file loaded.
       370 I: CFG : Using wifi settings from EEPROM.
[   384][E][vfs_api.cpp:105] open(): /littlefs/drd.dat does not exist, no permits for creation
       405 I: GYRO: Detected MPU6050/MPU6000.
      1308 I: GYRO: Movement detected (500)	562	204	332.
      1310 I: Main: Battery 0.00 V, Gyro=0.00, Run-mode=1.
      1413 I: WIFI: Connecting to wifi using stored settings xxxxxx.
..
      1727 I: WIFI: Connected to wifi vodafoneEBC8 ip=xxx.xxx.xxx.xx.
      1727 I: WIFI: Using mDNS name brewpill.

```
and then when I load the webui

```
     59117 I: WEB : webServer callback for /api/status(get).
[ 59125][E][vfs_api.cpp:105] open(): /littlefs/gravitymon.json does not exist, no permits for creation
     59152 I: WEB : webServer callback for /favicon.ico (Memory).
     59170 I: WEB : webServer callback for /api/config(read).
     59196 I: GYRO: Movement detected (500)	558	201	331.
     59211 I: WEB : webServer callback for /api/config/format(read).
[ 59214][E][vfs_api.cpp:105] open(): /littlefs/http-1.tpl does not exist, no permits for creation
[ 59229][E][vfs_api.cpp:105] open(): /littlefs/http-2.tpl does not exist, no permits for creation
[ 59241][E][vfs_api.cpp:105] open(): /littlefs/http-3.tpl does not exist, no permits for creation
[ 59253][E][vfs_api.cpp:105] open(): /littlefs/influxdb.tpl does not exist, no permits for creation
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (async_tcp) 
Core  1 register dump:
PC      : 0x4009a817  PS      : 0x00060c36  A0      : 0x8009a2ac  A1      : 0x3ffdb9c0  
A2      : 0x3ffbf808  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060c23  
A6      : 0x00060c20  A7      : 0x0000abab  A8      : 0xb33fffff  A9      : 0xffffffff  
A10     : 0x00000001  A11     : 0x3ffb6ce4  A12     : 0x3ffb6c98  A13     : 0x00000000  
A14     : 0x007bf808  A15     : 0x003fffff  SAR     : 0x0000001e  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x40092efc  LEND    : 0x40092f07  LCOUNT  : 0x00000000  

Backtrace: 0x4009a814:0x3ffdb9c0 0x4009a2a9:0x3ffdba00 0x400832ec:0x3ffdba20 0x40083179:0x3ffdba50 0x40083183:0x3ffdba70 0x40091c0d:0x3ffdba90 0x40082d12:0x3ffdbab0 0x4011828f:0x3ffdbaf0 0x40147462:0x3ffdbb20 0x40147735:0x3ffdbb50 0x4014888e:0x3ffdbb90 0x40148f5e:0x3ffdbc30 0x4014b0ed:0x3ffdbc90 0x401468c6:0x3ffdbce0 0x4011e7bd:0x3ffdbe10 0x40092bed:0x3ffdbe30 0x40092c59:0x3ffdbe90 0x4009366e:0x3ffdbec0 0x401c3d52:0x3ffdbee0 0x401c3de5:0x3ffdbf20 0x401b8e4b:0x3ffdbf40 0x400e7457:0x3ffdbf60 0x400df3df:0x3ffdbf80 0x400df875:0x3ffdc500 0x401d8471:0x3ffdc5b0 0x400ed4da:0x3ffdc5d0 0x401d8987:0x3ffdc610 0x400e8e6d:0x3ffdc630 0x400e8b92:0x3ffdc650 0x400e8d02:0x3ffdc670 0x400e8ebe:0x3ffdc700 0x400e8b92:0x3ffdc740 0x400e8d02:0x3ffdc760 0x400e951e:0x3ffdc7f0 0x400ea44e:0x3ffdc830 0x400eb10d:0x3ffdc870 0x400eb309:0x3ffdc8a0 0x401b7ca1:0x3ffdc8c0 0x401b7d19:0x3ffdc8f0 0x401b87e6:0x3ffdc910

ELF file SHA256: 52eb6a87ad513353
```
and then back to "normal" output:

```
      5679 I: WEB : Configuring web server.
      5701 I: WEB : File=error.log, 910 bytes
      5716 I: WEB : File=gravitymon2.json, 2060 bytes
      5731 I: WEB : File=mqtt.tpl, 1336 bytes
      5742 I: WEB : File=reset.dat, 1 bytes
      5743 I: WEB : Setting up async web handlers.
      5747 I: WEB : Web server started.
      5758 I: WEB : Setting up handlers for web server.
      5761 I: WEB : Web server started.
      5762 I: WS  : Starting serial websocket
      5762 I: SDBG: Serial logging started at 115200.
      5773 I: Main: Setup completed.
      6290 I: WEB : webServer callback for /api/ping.
      6663 I: GYRO: Movement detected (500)	560	202	331.
      7561 I: GYRO: Movement detected (500)	558	203	331.
      8459 I: GYRO: Movement detected (500)	558	199	330.
      9357 I: GYRO: Movement detected (500)	560	199	331.
     10255 I: GYRO: Movement detected (500)	559	202	329.
     11154 I: GYRO: Movement detected (500)	560	202	330.
     12052 I: GYRO: Movement detected (500)	559	200	330.
     12950 I: GYRO: Movement detected (500)	559	199	331.
     13046 I: WEB : webServer callback for /api/ping.
     13849 I: GYRO: Movement detected (500)	560	200	330.
     14747 I: GYRO: Movement detected (500)	560	201	330.
     15645 I: GYRO: Movement detected (500)	560	202	330.
     16543 I: GYRO: Movement detected (500)	560	202	330.
```
But everytime I tried to load the WebUI it crashes like the ouput before

GabrielPeral, 2025-07-15:
yes, I did configure the MQTT, more specifically for home assistant2, (is the idea to use it with HA) let me try it out without configure it

mp-se, 2025-07-15:
Try to configure it but if the same issue comes up again then its probably related to the file system and arduino. 

Did any of the older versions work?

GabrielPeral, 2025-07-16:
Perfect, thanks! So far it seems like that was the issue, using MQTT with homeasssistant2 format, I set to just home assistant format and seems like it is working for now all the time.
Just one last question and I will close the ticket, I saw that in the releases notes of 2.2.0 it is mentioned:

> Only ESP32! The voltage divider for the battery voltage has been changed since there where issues with some ESP32 boards. 220k + 220k 

That means that for my board, should I go with those values instead of 470K + 220K?

Thanks a lot for your help! I really appreciate it!

mp-se, 2025-07-16:
Yes